AllskyTeam/allsky #4034: [BUG] v2024.12.06 install failure - SUNWAIT repository not found

jcauthen78, 2024-12-09:
### Camera model

ASI178MC

### Operating System

Bullseye

### Allsky version

v2024.12.06

### Pi and amount of memory (512 MB, 1, 2, 4, or 8 GB)

8gb

### Bug Description

Fresh install.
RPI4, 8gb
VERSION="11 (bullseye)"

* Preparing Allsky commands.

*** ERROR:
**********
Compile failed
**********

* The full log file is in /home/pi/allsky/config/logs/make_all.log
The end of the file is:
	2024-12-08 17:04:08 Done.
	2024-12-08 17:04:08 Building startrails program...
	2024-12-08 17:04:19 Done.
	2024-12-08 17:04:19 Building keogram program...
	2024-12-08 17:04:31 Done.
	2024-12-08 17:04:31 Building sunwait...
	fatal: not a git repository (or any of the parent directories): .git

	make[1]: *** [Makefile:115: sunwait] Error 128
	make[1]: Leaving directory '/home/pi/allsky/src'
	make: *** [Makefile:24: all] Error 2

[install.log](https://github.com/user-attachments/files/18054687/install.log)
[make_all.log](https://github.com/user-attachments/files/18054689/make_all.log)

jcauthen78, 2024-12-09:
Attempted system Reboot & ./install.sh again. 
Fatal error still present.

EricClaeys, 2024-12-09:
@Alex-developer, can you take a look at this please?

@jcauthen78, Jim, what did you use for the `git clone ...` command?

EricClaeys, 2024-12-09:
@jcauthen78, please run `ls -FCa ~/allsky/src/sunwait-src`

jcauthen78, 2024-12-09:
This was the install -

git clone --depth=1 --recursive --branch v2024.12.06 https://github.com/AllskyTeam/allsky.git

entirely possible its incorrect. 

Results of the LS: 
pi@allsky:~/allsky $ ls -FCa ~/allsky/src/sunwait-src
./            .git        makefile*  print.h*     sunriset.h  SunwaitPi.vcxproj  USAGE.txt
../           .gitignore  openwrt/   README.md    sunwait.c*  sunwait.sln
examples.txt  LICENSE     print.c*   sunriset.c*  sunwait.h*  sunwait.vcxproj

EricClaeys, 2024-12-09:
Jim, v2024.12.06 is official so the `--branch` option shouldn't be used.  That likely caused the problem.
Please use `git clone --depth=1 --recursive  https://github.com/AllskyTeam/allsky.git`

jcauthen78, 2024-12-09:
Understood & successful.
I'll try not to break my SD card in half again, destroying my core AllSky system 🤦‍♂️

Closing: Successful
Reason: User error

closed 2024-12-09 by jcauthen78

EricClaeys, 2024-12-09:
Easy enough fix.