AllskyTeam/allsky #4036: [FEATURE REQUEST] Add Statement for Camera Functionality Prior to Installation

chicago925, 2024-12-09:
### If this is a new feature, describe what you would like.

There should be a note in the Pre-Installation section of the documentation that the camera being used should be physically connected to the Raspberry Pi, libcamera must be enabled, and any other required changes needed to support the proper operation of the camera should be completed.

EricClaeys, 2024-12-09:
Thanks for the suggestion.  I added the following to the top of the "Allsky installation" page:

[image]

closed 2024-12-09 by Alex-developer

chicago925, 2024-12-09:
This should help others who are installing from scratch - thanks guys. Just a small change needed to the wording above; in the phrase "if the camera number..." the word "in" is spelled as "n". Please correct.

EricClaeys, 2024-12-09:
Fixed.  Thanks for reporting

PR #4040 (2024-12-09): Mention camera must be connected prior to installation
Fixes #4036