AllskyTeam/allsky #4038: I just upgraded my setup and it went smoothly.  The only thing I've noticed is when I was playing with the new allsky-config command line tool.  When I ask it to show connected cameras, it says there are no cameras connected to the Pi.  I have a Pi HQ camera, and the allsky website is using it just fine, so something is going on with that command.

EricClaeys, 2024-12-09:
I just upgraded my setup and it went smoothly.  The only thing I've noticed is when I was playing with the new allsky-config command line tool.  When I ask it to show connected cameras, it says there are no cameras connected to the Pi.  I have a Pi HQ camera, and the allsky website is using it just fine, so something is going on with that command.

_Originally posted by @ranous in https://github.com/AllskyTeam/allsky/discussions/3986#discussioncomment-11503578_

EricClaeys, 2024-12-09:
@ranous, thanks for reporting.  I just tried and it's working fine:
[image]

Please run:
~~~
allsky-config --debug show_connected_cameras
libcamera-still --list-cameras
cat ~/allsky/config/connected_cameras.txt
~~~

Thanks

ranous, 2024-12-09:
Here's the output of those commands:

ranous@allsky:~ $ allsky-config --debug show_connected_cameras

Executing: show_connected_cameras :

There are no cameras connected to the Pi.
ranous@allsky:~ $ libcamera-still --list-cameras
Available cameras
-----------------
0 : imx477 [4056x3040] (/base/soc/i2c0mux/i2c@1/imx477@1a)
    Modes: 'SBGGR10_CSI2P' : 1332x990 [30.00 fps - (0, 0)/0x0 crop]
           'SBGGR12_CSI2P' : 2028x1080 [30.00 fps - (0, 0)/0x0 crop]
                             2028x1520 [30.00 fps - (0, 0)/0x0 crop]
                             4056x3040 [30.00 fps - (0, 0)/0x0 crop]

ranous@allsky:~ $ cat ~/allsky/config/connected_cameras.txt
ranous@allsky:~ $

EricClaeys, 2024-12-09:
Interesting.  The camera's clearly connected by Allsky doesn't see it.
Please run the following:
~~~
cd ~/allsky/scripts
source ../variables.sh
source functions.sh
set -x; get_connected_cameras_info true; set +x
~~~
Please do a screenshot of the output if you can, rather than copy/paste, then attach the screenshot.

FYI, here's what I get on my Pi with 2 RPi cameras and one ZWO camera:
[image]

ranous, 2024-12-09:
Ok, here's the output.  

[image]

EricClaeys, 2024-12-09:
Thanks, that helps.  The `libcamera-still` command is failing, possibly because Allsky is running.
Please run `libcamera-still  --timeout 1 --nopreview; echo -e "\n===== ret=$?"`
If that fails or you do NOT get `ret=0`, stop Allsky, rerun the command, then restart Allsky. 
Thanks for helping to fix this.

ranous, 2024-12-09:
Ok, that last bit seemed to fix it.  If I run allsky-config now, it shows my camera when I list connected cameras.

ranous, 2024-12-10:
Running that last test was successful, ending in ret=0.  I reran allsky-config, and it displayed the camera.  That said, it might have been something else.  I didn't try rerunning allsky-config after those steps, so it might have been something earlier.  I don't think I rebooted after updating allsky.  I certainly didn't reboot after allsky-config failed to find the camera.  I did updated the configuration in Allsky Settings which restarted allsky, so maybe that did it.

EricClaeys, 2024-12-10:
@ranous, Alex, so everything's working now, right?
If so, would you please close this Issue?
Thanks - Eric

closed 2024-12-12 by EricClaeys