AllskyTeam/allsky #4047: [BUG] Remote Website - missing 'viewSettings/allskySettings.php' file

jcauthen78, 2024-12-10:
### Camera model

ASI178MC

### Operating System

Bullseye

### Allsky version

v2024.12.06

### Pi and amount of memory (512 MB, 1, 2, 4, or 8 GB)

RPI 4b, 4gb

### Bug Description

Reinstalling remote website. Allsky completely reinstalled 2 days ago after I broke the original MicroSD card in half 🤦‍♂️.
Renamed old allsky folder on remote site (to preserve startrails, keograms, and timelapses). 
Followed docs to create new allsky folder and required subfolders. 
ran: `./remoteWebsiteInstall.sh` and it shows as **successful**
manually ran: `./scripts/postData.sh` for the initial data push, **successful**. 
Enabled remote website _(not server)_ in settings, filled in all the information and saved. 
Edited the `remote_configuration.json` and saved. 
On the remote website (https://jimsnerdstuff/allsky) I can click the info button, and it works. Within the info that comes up, the `Click to view allsky settings link` appears to be broken. 
a page comes up with the following text:
```
This Allsky Website is not fully configured so its settings cannot be displayed.
It is missing the 'viewSettings/allskySettings.php' file.

To fix: make sure the Website is enabled in the Websites and Remote Server Settingssection of the WebUI's Allsky Settings page.
```
_(of note, small typo in this page - **Settingssection** needs a space added to it)_

From what I can see, the website is configured correctly in the WebUI. In FileZilla, the `viewSettings` folder contains `README.txt` and `settings.json`, and nothing else. The attached remote config file looks correct, I'm not seeing any bad/missing edits. The attached settings.json looks correct too, the website is enabled ("useremotewebsite": true,). I've also attached a snapshot of my allsky.log with debug4 enabled.

If I've missed any needed troubleshooting files, please let me know. Looking through the Issues & discussions, I can't find something similar to this issue. 

[allsky_log.txt](https://github.com/user-attachments/files/18084643/allsky_log.txt)
[remote_configuration_json.txt](https://github.com/user-attachments/files/18084687/remote_configuration_json.txt)
[settings_json.txt](https://github.com/user-attachments/files/18084689/settings_json.txt)
[configuration_json.txt](https://github.com/user-attachments/files/18084690/configuration_json.txt)
[install_log.txt](https://github.com/user-attachments/files/18084787/install_log.txt)
[remoteWebsiteInstall_log.txt](https://github.com/user-attachments/files/18084812/remoteWebsiteInstall_log.txt)

EricClaeys, 2024-12-10:
@jcauthen78, Jim,
Very good, detailed report.  Thanks.
Someone reported the same missing file a couple days ago but for their local Website.  I  had a brain fart and didn't think of fixing it for remote Websites.  I will fix it and the misspelling.  If you run `postData.sh --allfiles` (which is what the installation should have done), it will upload the missing file.

> Renamed old allsky folder on remote site (to preserve startrails, keograms, and timelapses).  Followed docs to create new allsky folder and required subfolders.

Do you remember exactly what documentation told you to do that?  I need to fix it since with this release all you need to do is have an `allsky` folder (even it it has all the old files) and a working connection to the server and `remote_website_install.sh` will do everything else - copy all the new source files, remove any old unneeded files, and leaves any startrails, keograms, and timelapse already there.  It also enables the remote website.  When done, if it's a new Website you need to update `remote_configuration.json`.

jcauthen78, 2024-12-10:
`./scripts/postData.sh --allfiles` was successful & file is now on the website. Is this what the brain-fart was? or the text fix? Just wanting to understand if it was user error on my part.

Re the documentation question - just dug through it all again and realized i must have clicked on `Remote Server` instead of `Allsky Website` - user error on creating the new folders. Glad to know the installer takes care of all of this :)

The renaming of the older allsky folder was my choice as a just-in-case. The installer wasn't clear if the startrails, keograms, or startrails were retained or not, so leaned on the side of ensuring retention (paranoia lol). It mentioned overwriting files, but didn't mention any retention. Maybe something to clarify within the installer? 

[image]

EricClaeys, 2024-12-11:
@jcauthen78, my brain fart was fixing the missing file for local Websites but not realizing it was also a problem for remote Websites.
Doing all the extra work was "user error".  I just looked at the `Remote Server` documentation page but couldn't see a good way to change it to avoid someone else from making the same mistake.  If you can think of something, let me know.

I agree the installer needs to say it will NOT overwrite any startrails, keograms, or timelapse on the remote Website.  Good catch!
Just created [Issue #4052](https://github.com/AllskyTeam/allsky/issues/4052) for that.

jcauthen78, 2024-12-11:
Looking through the other docs, I came across `http://allsky.local/documentation/explanations/SSL.html`. Part of the Requirements section has a big red tab/header/note/thing explaining a little more about the SLL stuff.  

What about something similar for the `Remote Server` page at the top that states something like "**Jim Reminds You - This Is Not For Your AllSky Website!**"  usually something big in red is eye-catching. lol.  Wording to be set more professionally ofcourse ;)  Since the CSS (i think) has that formatting in it already, the change shouldn't break the existing documentation??  

[image]

EricClaeys, 2024-12-11:
Good idea.  It would be unprofessional not to include your last name!!

The red box is for warnings and the similar yellow box is for info we want to highlight.  The CSS includes both.

Wanna try making the change yourself?
`~/allsky/html/documentation/installations/remoteServer.html` (or something similar).
~~~
<blockquote class="warning">
Jim C. says ...
</blockquote>
~~~

You can then look at it on your Pi.  Attach the file when done.

jcauthen78, 2024-12-11:
Very nice! That made for an easy change. 

[image]
[RemoteServer_html.txt](https://github.com/user-attachments/files/18089427/RemoteServer_html.txt)

PR #4062 (2024-12-12): Update RemoteServer.html: warning about what page is about
Fixes #4047
commit 78562935 (2024-12-12): Update RemoteServer.html: warning about what page is about

Fixes #4047
closed 2024-12-12 by EricClaeys