AllskyTeam/allsky #4063: [BUG] Allsky sftp setting for Remote Website does not treat private key entry properly

chicago925, 2024-12-12:
### Camera model

Arducam 16MP

### Operating System

Bookworm

### Allsky version

Version: v2024.12.06

### Pi and amount of memory (512 MB, 1, 2, 4, or 8 GB)

Raspberry PI Model 4B (4 GB)

### Bug Description

Setting sftp protocol in Allsky for file transfers to a VM in Google Cloud via the Remote Website area in Allsky Settings does not work. Google Cloud requires private/public keys for sftp. When the Allsky test upload is sent the process fails:

Test upload to Remote Website FAILED with RET=99.
* Unable to login for unknown reason.
  FIX: Make sure the port is correct and your network is working.
       On your Pi, run:  ssh @
       When prompted to enter 'yes' or 'no', enter 'yes'.
       You may need to do this if the IP address of your Pi changed.

=================== RAW OUTPUT:
    
    Commands used are in: /home/brian/allsky/tmp/lftp_cmds/testUpload.sh.txt
    Upload output from '/tmp/testUpload.sh.txt:
       *** upload.sh: ERROR, RET=99:
    FILE_TO_UPLOAD='/tmp/testUpload.sh.txt'
    REMOTE_HOST='leshintech.click'
    DIRECTORY='allsky/'
    TEMP_NAME='testUpload.sh-19451'
    DESTINATION_NAME='testUpload.sh.txt'
    
    <--- allskytest@leshintech.click: Permission denied (publickey).
    **** pty read: pseudo-tty: Input/output error
    **** allskytest@leshintech.click: Permission denied (publickey).
    <--- allskytest@leshintech.click: Permission denied (publickey).
    **** pty read: pseudo-tty: Input/output error
    **** allskytest@leshintech.click: Permission denied (publickey).
    cd: Fatal error: max-retries exceeded (allskytest@leshintech.click: Permission denied (publickey).)
       (December 12, 04:17:58 PM)

The same credentials work from the sftp command line in the Pi OS for transfers to the Google Cloud VM, as does a connection from FIlezilla from a local desktop. Log files and image attached below:

[image]
[settings.json](https://github.com/user-attachments/files/18118272/settings.json)
[upload_errors.txt](https://github.com/user-attachments/files/18118273/upload_errors.txt)
[configuration.json](https://github.com/user-attachments/files/18118274/configuration.json)
[allsky.log](https://github.com/user-attachments/files/18118275/allsky.log)

Alex-developer, 2024-12-13:
Thanks for the report we will investigate

Alex-developer, 2024-12-14:
I have setup a VM in the Google cloud and tested the image uploading which all works fine.

Few things to check

1) In the screenshot above you have a password filled in - Thats not neeed
2) Whats in the file you used for the 'path to the keys'? The key is normally available in ~/.ssh/id_rsa.pub
3) What access method are you using or accessing the VM? I am using managed keys rather than os_login
4) Your image directory doesn't look right unless you have installed the website in the 'allskytest' home directory, 

Can you ssh to the server from the pi?

[image]

[image]

chicago925, 2024-12-14:
In answer to these questions:

In the screenshot above you have a password filled in - Thats not neeed
  > I removed that in later testing and it made no difference.

Whats in the file you used for the 'path to the keys'? The key is normally available in ~/.ssh/id_rsa.pub
 > I generated the keys to reside in ~allsky. The path to the private key was used here and I added the public key to the VM.

What access method are you using or accessing the VM? I am using managed keys rather than os_login.
 > I am using managed keys as well.

 Your image directory doesn't look right unless you have installed the website in the 'allskytest' home directory,
> Yes - it is not correct. I am using ftp for now to get the files up on the website. I have changed the home directory for the allskytest user to /var/www/html and uploaded the files using remoteWebsiteInstall.sh into a directory in html called allsky. I have the image directory set to 'allsky'. However the files and directories are coming across from the Pi and creating 600 permissions on the remote server so I need to see what is going on.  Also the images are getting uploaded directly into the allsky directory and not into a separate  image directory as on the Pi. If I change the image directory on the Pi to 'images' and use the remoteWebInstall.sh script the website gets loaded into that directory so the remote website is not showing any images. I know this is not this issue but I will need help with this too.

Finally I can ssh to the server from the Pi (URL/IP's removed):

brian@allsky-test:~/allsky $ ssh -i /home/brian/keys brian@------------
Linux bdlallsky 6.1.0-28-cloud-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.1.119-1 (2024-11-22) x86_64

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Sat Dec 14 17:07:21 2024 from -----------
brian@bdlallsky:~$

chicago925, 2024-12-14:
Also, what do you have for image directory - that may help me with my problem around where the images get stored.

Alex-developer, 2024-12-14:
Thanks for the responses.

I setup the allsky site in the users (pi) home directory and setup the apache userdir module to serve the site. So my path is ~/allsky/public_html

I am wondering if its related to the way you have setup the keys? Could you try using 'standard' ssh keys i.e. in the .ssh directory for your user so that you do not have to use the -i option on ssh

Thanks

Alex

chicago925, 2024-12-14:
Thanks for all your help on this Alex. I'll try using the keys in .ssh and let you know the result.

Another question: Where do the images get uploaded to the remote VM? Mine are getting uploaded into the allsky directory and I don't see an images directory at all.

Alex-developer, 2024-12-14:
You would have to specify it in the main Allsky settings. If you look in my screenshot I set it to '/home/pi/public_html' as thats where I setup the remote website on the VM

Alex

chicago925, 2024-12-14:
Sorry. I don't see the images directory in your directory listing above. Where is it? Does it get created by the script and if so where is it created?

Alex-developer, 2024-12-14:
I am putting the image in the root folder of the website not in a separate directory, If you look in the list there is a file 'image.jpg' and in the settings screenshot the 'Image Directory' points to the public_html folder

Alex

chicago925, 2024-12-14:
OK - so do all of the uploaded images just pile up in that directory?

Alex-developer, 2024-12-14:
No its only uploading the last captured image, for startrails, keograms and timelapses are uploaded to the relevant folder

chicago925, 2024-12-14:
I'm seeing the image files piling up in mine. Something strange is going on. I'm going to reload the remote server.

Alex-developer, 2024-12-14:
> I'm seeing the image files piling up in mine. Something strange is going on. I'm going to reload the remote server.

Can you do a quick screenshot of the files before you re install it please

Thanks

Alex

chicago925, 2024-12-14:
Sorry - didn't but when I re-installed it did the same thing. Here is the screenshot
[image]

Alex-developer, 2024-12-14:
Can you check /var/log/allsky.log and see if there are any errors looks like it cant remove a file

chicago925, 2024-12-14:
No errors in the log for the last hour

Alex-developer, 2024-12-14:
So just a quick summary.

1) Is the ftp issue sorted? You seem to have images now
2) @EricClaeys have you seen the image issue before?

Lets see if Eric has seen this

Thanks

Alex

chicago925, 2024-12-14:
The sfto issue is still open. I need to retest using the key files in the .ssh folder as opposed to the ones I used with the -i option.

The images are just piling up. I won't open a bug yet since it may be something I did wrong.

Alex-developer, 2024-12-14:
> The sfto issue is still open. I need to retest using the key files in the .ssh folder as opposed to the ones I used with the -i option.
> 
> The images are just piling up. I won't open a bug yet since it may be something I did wrong.

Sorry I am a bit confused

If the ftp isn't working how are the images getting onto the server?

Thanks

Alex

chicago925, 2024-12-14:
Just plain ftp is working sftp (using the keys) is not.

Alex-developer, 2024-12-14:
> Just plain ftp is working sftp (using the keys) is not.

Ah that makes sense :-)

lets see if Eric has any ideas, I'll have a difg thru the code but he may have seen this before

Thanks

Alex

chicago925, 2024-12-14:
Cool - and just to make sure I am using the git clone from 12/7 so if the remote install or upload scipts have changes since then I can get whatever is needed and try again

Alex-developer, 2024-12-14:
Just to be safe I would do another git clone of Allsky. There have been some changes, they should not affect this but you never know !

Alex

EricClaeys, 2024-12-14:
@chicago925, I suspect you have this setting enabled:
[image]

This causes your images to have the YYYYMMDDHHMMSS in them, which doesn't work for an Allsky Website which only ever looks at the most recent image called `image.jpg`.

By design, an Allsky Website only has the most recent `image.jpg` file and multiple startrails, keograms, and timelapse.
Eric

chicago925, 2024-12-15:
That was indeed the issue Eric. Thanks!!

Now to clean up the broek image files and back to the sftp testing....

Alex-developer, 2024-12-15:
> That was indeed the issue Eric. Thanks!!
> 
> Now to clean up the broek image files and back to the sftp testing....

I knew @EricClaeys would know :-)

Please let me know how you get on with the sftp. I am pretty confident its just down to the way the keys are setup

Alex

chicago925, 2024-12-16:
The issue was the use of a local public key file that was not in the .ssh directory and not named id_rsa.pub per the ssh generation defaults. Once I regenerated the keys and pointed the SSH Key File to /home/brian/.ssh/id_rsa.pub I was able to connect with sftp.

Thank you Alex and Eric for the assistance for this problem and the remote image problem.  I am now closing this issue as resolved.

closed 2024-12-16 by chicago925

EricClaeys, 2024-12-16:
@chicago925, is there anything we can do to Allsky or its documentation to decrease the chance of this happening to someone else?

Eric

chicago925, 2024-12-16:
Hmmm - maybe add something in the doc to the effect that you should use 'ssh-keygen' to generate the keys and when prompted to choose the location for the key pair press enter to save the key pair in the default location (~/.ssh/id_rsa).

Also in the actual Allsky Settings section for the Remote Website consider changing the entry name for the key from "SSH Key File" to "SSH Public Key File" and the note under the field to read "Path on the Pi to the SSH public key file"

EricClaeys, 2024-12-16:
@chicago925, thanks for the ideas. I added [Issue #4076](https://github.com/AllskyTeam/allsky/issues/4076) to make the changes in the first point release.