AllskyTeam/allsky #4076: sftp documentation clarifications

EricClaeys, 2024-12-16:
Hmmm - maybe add something in the doc to the effect that you should use 'ssh-keygen' to generate the keys and when prompted to choose the location for the key pair press enter to save the key pair in the default location (~/.ssh/id_rsa).

Also in the actual Allsky Settings section for the Remote Website consider changing the entry name for the key from "SSH Key File" to "SSH Public Key File" and the note under the field to read "Path on the Pi to the SSH public key file"

_Originally posted by @chicago925 in https://github.com/AllskyTeam/allsky/issues/4063#issuecomment-2544542931_

EricClaeys, 2025-02-03:
@chicago925,
I added "public" per your suggestion, both on the Settings page and in the documentation, as below

Regarding this:
> Add something in the doc to the effect that you should use 'ssh-keygen' to generate the keys and when prompted to choose the location for the key pair press enter to save the key pair in the default location (~/.ssh/id_rsa).

I think the documentation says that.  Do you agree?

[image]

chicago925, 2025-02-03:
Agreed! Thanks Eric.

PR #4280 (2025-02-03): Update allsky.html: Add "public" to SSH key file
Fixes #4076
closed 2025-02-08 by EricClaeys