AllskyTeam/allsky #4084: remoteWebsiteInstall.sh doesn't handle remote Websites without a config file

EricClaeys, 2024-12-18:
### Camera model

x

### Operating System

Buster

### Allsky version

x

### Pi and amount of memory (512 MB, 1, 2, 4, or 8 GB)

x

### Bug Description

x

Alex-developer, 2024-12-19:
Excluding existing images and videos keeps the */index.php files from being updated. - FIXED

Alex-developer, 2024-12-20:
Fix for index.php files added

TEST NOTES:
- All tests done using the dialog based installer

1) Fresh install of Allsky and Remote website
- All files uploaded OK
- Correctly displays warning that configuration.json needs updating
- Remote config updated on local pi and uploaded ok
- Displays warning that data is missing from data.json
- Ran postData.json
- Site running ok

2) Fresh install of Allsky with existing remote website
- NOTES:
 - The existing remote website is a copy of the v2024.12.06 website with version and README.md files added. 
 - Images andded to the Keogram, timelapse and startails folder
 - index.php in startrails, keograms and videos modified (to help check its been updated)
 - No remote config on the local pi
 - config.js on the website

- All files uploaded OK
- Correctly displays warning that configuration.json needs updating and that a very old website had been found
- All index.php files updated with new versions
- version, README.md and config.js deleted
- As expected remote website complains the configuration.json needs updating 
- Remote config updated on local pi and uploaded ok
- Displays warning that data is missing from data.json
- Ran postData.json
- Site running ok, all hitorical images still accesible

Alex-developer, 2024-12-20:
Fix added for the following

```
	if [[ -f ${ALLSKY_REMOTE_WEBSITE_CONFIGURATION_FILE} ]]; then
		# 1a.
		DT="FOUND"
		MSG="Found ${ALLSKY_REMOTE_WEBSITE_CONFIGURATION_FILE}."
		display_msg --logonly info "${MSG}"
		HAVE_LOCAL_CONFIG="true"

### FIX: ALEX: I don't think this "elif" part should be used.
# During Allsky upgrades, if the OLD directory exists users are asked if
# it should be used.  If "yes", then the prior remote Website config file was
# copied to the new Allsky, so 1a should match.
# If the user answered "no", don't use OLD Allsky, we probably shouldn't either.
# HAVE_PRIOR_CONFIG should also be deleted.
	elif [[ -f ${PRIOR_REMOTE_WEBSITE_CONFIGURATION_FILE} ]]; then
		# 1b.
		DT="FOUND ${PRIOR_REMOTE_WEBSITE_CONFIGURATION_FILE}."
		MSG="Found ${PRIOR_REMOTE_WEBSITE_CONFIGURATION_FILE}."
		display_msg --logonly info "${MSG}"
		HAVE_PRIOR_CONFIG="true"

	else
		DT="NOT FOUND"
		display_msg --logonly info "No local config file found."
	fi
```

Alex-developer, 2024-12-20:
All changes have been made - this now requires some testing

PR #4096 (2024-12-21): remoteWebsiteInstall.sh fixes
Fixes #4084

Several fixes and enhancements to remoteWebsiteInstall.sh.  See each commit for details.
closed 2024-12-21 by Alex-developer