AllskyTeam/allsky #4109: [BUG] allsky.logrotate.repo fails on Bookworm

EricClaeys, 2024-12-23:
### Camera model

x

### Operating System

Buster

### Allsky version

x

### Pi and amount of memory (512 MB, 1, 2, 4, or 8 GB)

x

### Bug Description

x

wwqwq2313, 2024-12-23:
Hello, we tried to solve the issue.

This is what we did:

Modified the postrotate scripts in the allsky.logrotate.repo file to use systemctl instead of invoke-rc.d for rotating rsyslog. This change addresses the issue where the rsyslog init script is not found on systems using systemd, such as Debian Bookworm.

You can review changes in this commit: https://github.com/wwqwq2313/AllskyTeam-allsky-4109/commit/f16134da6b8d3a3347126f56dcb58a79c3a81686.

> [!CAUTION]
> Disclaimer: The concept of solution was created by AI and you should never copy paste this code before you check the correctness of generated code. Solution might not be complete, you should use this code as an inspiration only.

---

Latta AI seeks to solve problems in open source projects as part of its mission to support developers around the world. Learn more about our mission at https://latta.ai/ourmission . If you no longer want Latta AI to attempt solving issues on your repository, you can block this account.

EricClaeys, 2024-12-23:
Other scripts have this:
~~~
	/usr/lib/rsyslog/rsyslog-rotate
~~~
whereas allsky.logrotate.repo has:
~~~
	invoke-rc.d rsyslog rotate > /dev/null
~~~

The AI response above has:
~~~
	systemctl kill -s HUP rsyslog.service
~~~

Alex-developer, 2024-12-26:
I have added a couple of fixes to my Allsky and will test them over the next few days before committing them

Alex-developer, 2024-12-27:
Changed logrotate config

Tested over a few days on my pi and is fine

To test this you must set the logs to rotate daily otherwise you will have to wait a week to see if its worked !

Daily rotation

```
/var/log/allsky.log
{
	rotate 2
	daily
	missingok
	notifempty
	compress
	delaycompress
	sharedscripts
	postrotate
		systemctl kill -s HUP rsyslog.service
	endscript
}

/var/log/allskyperiodic.log
{
	rotate 2
	daily
	missingok
	notifempty
	compress
	delaycompress
	sharedscripts
	postrotate
		systemctl kill -s HUP rsyslog.service
	endscript
}
```

closed 2025-02-08 by Alex-developer