AllskyTeam/allsky #4111: Allsky Website sort order doesn't work

EricClaeys, 2024-12-23:
Yes, I updated the thumbnail sort order in that .json, and it doesn't appear to update the sort order on either the local website on the pi or the remote website. (I updated both config json files for each website).

My website is here: https://bw-reka.synology.me/~allsky/

_Originally posted by @brianboru82 in https://github.com/AllskyTeam/allsky/discussions/4092#discussioncomment-11644591_

EricClaeys, 2024-12-23:
Also need to fix so the message about which sort order is used appears at the top of the page.

EricClaeys, 2024-12-23:
This will be fixed in the first point release, due ~ February, 2025.
In the meantime, if you copy the files in the attached zip file to `~/allsky/html/allsky` you will have the fix.

[sortOrderFix.zip](https://github.com/user-attachments/files/18225986/sortOrderFix.zip)

brianboru82, 2024-12-25:
Thanks, I'll try it out!

brianboru82, 2024-12-25:
That fix does work. Thank you very much!

PR #4113 (2024-12-23): 4111   fix Allsky Website sort order
Fixes #4111
closed 2025-02-08 by EricClaeys