AllskyTeam/allsky #4127: [BUG]

chvvkumar, 2024-12-28:
### Camera model

RPi HQ Camera

### Operating System

Bookworm

### Allsky version

v2024.12.06

### Pi and amount of memory (512 MB, 1, 2, 4, or 8 GB)

Pi 5, 4GB

### Bug Description

Seeing this error in my notification:

The website is accessible using the URL for both the image and the website mentioned in the error message so not sure if it's the port number causing this.

```
postToMap.sh: ERROR: Image URL 'https://allsky.challa.co:1982/current/resized/image.jpg' does not appear to be a valid image.ERROR: Website URL 'https://allsky.challa.co:1982/allsky/index.php/' does not appear to be a valid web site.   (December 27, 06:39:22 AM)
```

[image]

[allsky.log](https://github.com/user-attachments/files/18264416/allsky.log)
[settings.json.txt](https://github.com/user-attachments/files/18264417/settings.json.txt)
[configuration.json.txt](https://github.com/user-attachments/files/18264418/configuration.json.txt)

EricClaeys, 2024-12-28:
@chvvkumar, Kumar, change your WebUI login and password ASAP.
I am on your WebUI page and if I was hacker I could control your Pi.

chvvkumar, 2024-12-28:
> @chvvkumar, Kumar, change your WebUI login and password ASAP. I am on your WebUI page and if I was hacker I could control your Pi.

Thanks for the head's up!  I changed it to test some things but changed it back now.

EricClaeys, 2024-12-28:
For the Website URL error, remove `index.php` from the URL in the WebUI.   Note in the red message it appended a `/` to the URL because it assumes the URL points to a directory.  I will look at changing this.

After making the change, please run:
~~~
postToMap.sh  --debug --force
~~~

I will look at the image URL error when I am on my PC.  In the meantime, I am curious why you have `resized` in the URL?  I assume you created an `allsky/tmp/resized/` directory and are somehow copying the current image there?

EricClaeys, 2024-12-28:
You can lower your `Debug Level` to 3.  Level 4 is rarely needed anymore and makes the log harder to read.

Also, you can disable `Use dark frames` because they aren't implemented yet on RPi cameras.  If you made some changes to get them to work in the prior release, note these messages in the log file:
~~~
*** darkSubtract.sh: WARNING: 'AS_TEMPERATURE_C' not set; continuing without dark subtraction.
~~~

chvvkumar, 2024-12-28:
That seems to have fixed it! Thank you!

```shell
pi@allskypi5:[~/allsky/scripts]: postToMap.sh  --debug --force

check_URL(https://allsky.challa.co:1982/allsky/, remotewebsiteurl, Website URL) RET=0:
HTTP/2 200
server: nginx
date: Sat, 28 Dec 2024 01:05:43 GMT
content-type: text/html; charset=UTF-8
x-content-type-options: nosniff
cache-control: max-age=31536001, immutable
accept-ranges: bytes

.

check_URL(https://allsky.challa.co:1982/current/resized/image.jpg, remotewebsiteimageurl, Image URL) RET=0:
HTTP/2 200
server: nginx
date: Sat, 28 Dec 2024 01:05:44 GMT
content-type: image/jpeg
content-length: 201158
etag: "4036338291"
last-modified: Sat, 28 Dec 2024 01:05:02 GMT
x-content-type-options: nosniff
cache-control: no-cache, proxy-revalidate
accept-ranges: bytes

.

postToMap.sh: Uploading map data.

Executing:
curl --silent --show-error -i -H 'Accept: application/json' -H 'Content-Type:application/json' --data '{
"location": "OFallon, MO",
"owner": "Kumar",
"latitude": "38.758331N",
"longitude": "90.689438W",
"website_url": "https://allsky.challa.co:1982/allsky/",
"image_url": "https://allsky.challa.co:1982/current/resized/image.jpg",
"camera": "RPi HQ",
"lens": "Fisheye",
"computer": "RPi 5 Model B Rev 1, 4 GB",
"allsky_version": "v2024.12.06",
"machine_id": "bf352f9be0e04948b56f17b00a27d47e"
}' https://www.thomasjacquin.com/allsky-map/postToMap.php

Returned:
HTTP/2 200
date: Sat, 28 Dec 2024 01:05:43 GMT
content-type: text/html; charset=UTF-8
content-length: 12
server: o2switch-PowerBoost-v3

UPDATED 1 5.

postToMap.sh: Map data UPDATED.  You can make 4 more updates today.

```

The resized image is what I normally use everywhere other than the webui. I use it on:

The desktop display I made:
https://github.com/AllskyTeam/allsky/discussions/3630

As an input to my ML cloud detection project I made:
https://github.com/chvvkumar/simpleCloudDetect

Both of those pull a new image every 30s from the AllSky Pi and sending a large ~2.5MB image is not necessary. So there is a morgify cron task that takes the latest image, resizes it to 750x750 and saves it to `/resized/` folder.

As for the debug level-4, I am checking my periodic log for the module I developed for allsky-modules:
https://github.com/AllskyTeam/allsky-modules/tree/dev/allsky_mqttsubscribe

And finally, I will go ahead and disable the dark frame subtraction, I managed to get it working a while ago but I don't think I need to enable it any more.

EricClaeys, 2024-12-28:
Is everything fixed now, or just the directory URL?

chvvkumar, 2024-12-28:
Everything is fixed now but I am actually not sure why that message popped up today. I have been running with the same /resized/ image URL for quite a while now. 

This is an issue that still happens though, which is why I thought it might be the port number in the URL causing it. The current image doesn't show up on the map if the address has a host:port in the URL.

https://github.com/AllskyTeam/allsky/discussions/3967

EricClaeys, 2024-12-28:
I am closing this Issue.
I created [Issue #4128](https://github.com/AllskyTeam/allsky/issues/4128) for the problem you had with `index.php` in the URL, and the Map not working with URLs with port numbers is already documented.

closed 2024-12-28 by EricClaeys