AllskyTeam/allsky #4161: remoteWebsiteInstall.sh needs to use REMOTEWEBSITE_LFTP_COMMANDS

EricClaeys, 2025-01-03:
### Discussed in https://github.com/AllskyTeam/allsky/discussions/4160

<div type='discussions-op-text'>

<sup>Originally posted by **rbotero1976** January  2, 2025</sup>

[image]

I read through previous discussions and the Troubleshooting guide and added the "set ssl:verify-certificate no" to my FTP settings here:
</div>

`remoteWebsiteInstall.sh` needs to add REMOTEWEBSITE_LFTP_COMMANDS to the lftp command to pick up any commands the user had to set.

EricClaeys, 2025-01-04:
This file should fix the problem.
Copy it to `~/allsky` then:
~~~
cd ~/allsky
# Copy the file here
mv remoteWebsiteInstall.sh remoteWebsiteInstall.sh-OLD
mv remoteWebsiteInstall.sh.txt remoteWebsiteInstall.sh
chmod 775 remoteWebsiteInstall.sh
./remoteWebsiteInstall.sh
~~~

[remoteWebsiteInstall.sh.txt](https://github.com/user-attachments/files/18305222/remoteWebsiteInstall.sh.txt)

rbotero1976, 2025-01-04:
Thank you Eric.  That seems to have done the trick.  The script ran fine and I updated just one XX_NEED_TO_UPDATE_XX in the Title line.

[image]

However, all my images in the remote website have changed their sorting order to ascending even though I have chosen descending in the WebUI configuration:

[image]

[image]

EricClaeys, 2025-01-05:
I'm glad that fixed the problem, thanks for letting me know.

The sort order not working is a [known Issue](https://github.com/AllskyTeam/allsky/issues/4111).  Kinda embarrassing - I introduce a new feature and it doesn't work.  Whoever tested that should be fired!
There's a fix at that link if you don't want to wait until February 1 when the point release comes out.

closed 2025-02-08 by EricClaeys