AllskyTeam/allsky #4163: [BUG] overlay problem - no iss / sun / moon data

newgecko, 2025-01-03:
### Camera model

RPI HQ

### Operating System

Bookworm

### Allsky version

2024.12.06

### Pi and amount of memory (512 MB, 1, 2, 4, or 8 GB)

RPi4 8GB

### Bug Description

Hello,

all the best for the new year. I have reinstalled my Raspi and my Allsky cam over the holidays. My Allsky version: v2024.12.06 does not get any data regarding sun/moonrise/sunset/ISS/HST, i.e. nothing is displayed as an overlay. Maybe someone can have a look at my debug logs and has an idea.

Best regards

**allskylog:**

2025-01-03T01:11:06.866364+01:00 allsky allsky[9061]:   > Focus: 12,241888
2025-01-03T01:11:06.921294+01:00 allsky allsky[9061]:   > Got:    shutter_us: 3,2 sec, gain: 1,000, mean: 0,217, target mean: 0,200, diff (target - mean): -0,017
2025-01-03T01:11:06.921466+01:00 allsky allsky[9061]:   > index: 1, meanHistory[]=0,232 exposureLevelHistory[]=60, newNean=0,232
2025-01-03T01:11:06.921524+01:00 allsky allsky[9061]:   > index: 2, meanHistory[]=0,184 exposureLevelHistory[]=60, newNean=0,600
2025-01-03T01:11:06.921583+01:00 allsky allsky[9061]:   > index: 0, meanHistory[]=0,217 exposureLevelHistory[]=60, newNean=1,251
2025-01-03T01:11:06.921634+01:00 allsky allsky[9061]:   > New mean target: 0,222, mean_forecast: 0,250, mean_diff (newMean - target mean): 0,022, idx=0, idxN1=2
2025-01-03T01:11:06.921687+01:00 allsky allsky[9061]:   > ExposureChange clipped to 3 (diff from last change: 0)
2025-01-03T01:11:06.921762+01:00 allsky allsky[9061]:   > ++++++++++ Prior image mean good - no changes needed, mean=0,217, target mean=0,200 threshold=0,100
2025-01-03T01:11:06.921822+01:00 allsky allsky[9061]:   > Saving NIGHT image 'image-20250103011100.jpg'
2025-01-03T01:11:06.923284+01:00 allsky allsky[9061]:   > Sleeping: 57,8 sec
2025-01-03T01:11:06.929020+01:00 allsky allsky[10264]: saveImage.sh NIGHT /home/pi/allsky/tmp/image-20250103011100.jpg DATE=20250103 TIME=011100 AUTOEXPOSURE=1 sAUTOEXPOSURE=(auto) EXPOSURE_US=3174802 sEXPOSURE=3,2 sec AUTOGAIN=1 sAUTOGAIN=(auto) GAIN=1 AUTOWB=1 sAUTOAWB=(auto) WBR=2,500000 WBB=2 MEAN=0,216755 BIN=1 FLIP=none BIT_DEPTH=0 FOCUS=12 DARKFRAME=0 eOVERLAY=1
2025-01-03T01:11:12.398932+01:00 allsky allsky[10298]: /usr/lib/python3/dist-packages/numpy/core/getlimits.py:518: UserWarning: The value of the smallest subnormal for <class 'numpy.float64'> type is zero.
2025-01-03T01:11:12.399156+01:00 allsky allsky[10298]:   setattr(self, word, getattr(machar, word).flat[0])
2025-01-03T01:11:12.399219+01:00 allsky allsky[10298]: /usr/lib/python3/dist-packages/numpy/core/getlimits.py:89: UserWarning: The value of the smallest subnormal for <class 'numpy.float64'> type is zero.
2025-01-03T01:11:12.399261+01:00 allsky allsky[10298]:   return self._float_to_str(self.smallest_subnormal)
2025-01-03T01:11:12.399302+01:00 allsky allsky[10298]: /usr/lib/python3/dist-packages/numpy/core/getlimits.py:518: UserWarning: The value of the smallest subnormal for <class 'numpy.float32'> type is zero.
2025-01-03T01:11:12.399341+01:00 allsky allsky[10298]:   setattr(self, word, getattr(machar, word).flat[0])
2025-01-03T01:11:12.399381+01:00 allsky allsky[10298]: /usr/lib/python3/dist-packages/numpy/core/getlimits.py:89: UserWarning: The value of the smallest subnormal for <class 'numpy.float32'> type is zero.
2025-01-03T01:11:12.399419+01:00 allsky allsky[10298]:   return self._float_to_str(self.smallest_subnormal)
2025-01-03T01:11:13.416808+01:00 allsky allsky[10298]: INFO: Loading /home/pi/allsky/config/settings.json
2025-01-03T01:11:13.417512+01:00 allsky allsky[10298]: INFO: ===== Running night flow...
2025-01-03T01:11:13.418067+01:00 allsky allsky[10298]: INFO: --------------- Running Module allsky_loadimage ---------------
2025-01-03T01:11:13.418685+01:00 allsky allsky[10298]: INFO: Image /home/pi/allsky/tmp/image-20250103011100.jpg Loaded
2025-01-03T01:11:13.418999+01:00 allsky allsky[10298]: INFO: --------------- Running Module allsky_cloud ---------------
2025-01-03T01:11:13.419045+01:00 allsky allsky[10298]: ERROR: Module cloud failed on line 255 - No I2C device at address: 0x5a
2025-01-03T01:11:13.419087+01:00 allsky allsky[10298]: INFO: Module allsky_temp.py disabled; ignoring.
2025-01-03T01:11:13.419126+01:00 allsky allsky[10298]: INFO: Module allsky_lightgraph.py disabled; ignoring.
2025-01-03T01:11:13.419166+01:00 allsky allsky[10298]: INFO: --------------- Running Module allsky_clearsky ---------------
2025-01-03T01:11:13.419206+01:00 allsky allsky[10298]: INFO: Created star template. Radius - 6
2025-01-03T01:11:13.419298+01:00 allsky allsky[10298]: INFO: Sky is clear. 110 stars found, clear limit is 10
2025-01-03T01:11:13.419360+01:00 allsky allsky[10298]: INFO: MQTT disabled
2025-01-03T01:11:13.419413+01:00 allsky allsky[10298]: INFO: --------------- Running Module allsky_ai ---------------
2025-01-03T01:11:13.419453+01:00 allsky allsky[10298]: ERROR: Camera type not set, check AllSkyAI settings...
2025-01-03T01:11:13.419494+01:00 allsky allsky[10298]: INFO: Module allsky_maskimage.py disabled; ignoring.
2025-01-03T01:11:13.419533+01:00 allsky allsky[10298]: INFO: Module allsky_boilerplate.py disabled; ignoring.
2025-01-03T01:11:13.419572+01:00 allsky allsky[10298]: INFO: --------------- Running Module allsky_starcount ---------------
2025-01-03T01:11:13.419628+01:00 allsky allsky[10298]: INFO: Total stars found 36
2025-01-03T01:11:13.419679+01:00 allsky allsky[10298]: INFO: --------------- Running Module allsky_saveintermediateimage ---------------
2025-01-03T01:11:13.419724+01:00 allsky allsky[10298]: INFO: Image /home/pi/allsky/images/20250103-clean/image-20250103011100.jpg Saved
2025-01-03T01:11:13.419774+01:00 allsky allsky[10298]: INFO: --------------- Running Module allsky_script ---------------
2025-01-03T01:11:13.419814+01:00 allsky allsky[10298]: ERROR: cannot access /home/pi/allsky/custom_scripts
2025-01-03T01:11:13.419860+01:00 allsky allsky[10298]: INFO: --------------- Running Module allsky_meteor ---------------
2025-01-03T01:11:13.419901+01:00 allsky allsky[10298]: INFO: 0 Meteors found, 0 Lines detected
2025-01-03T01:11:13.419955+01:00 allsky allsky[10298]: INFO: --------------- Running Module allsky_export ---------------
[... 431 lines cut ...]
    "useremoteserver": false,
    "remoteserverimagedir": "",
    "remoteserverprotocol": "ftps",
    "remoteserverimageuploadoriginalname": false,
    "remoteservervideodestinationname": "",
    "remoteserverkeogramdestinationname": "",
    "remoteserverstartrailsdestinationname": "",
    "showonmap": true,
    "location": "Bernau,GER",
    "owner": "Nico Geisler",
    "camera": "RPi HQ",
    "lens": 2.1,
    "computer": "RPi 4 Model B Rev 1.4, 8 GB",
    "imagessortorder": "ascending",
    "showupdatedmessage": true,
    "uselogin": true,
    "notificationimages": true,
    "webuidatafiles": "",
    "daytimeoverlay": "overlay1-RPi_HQ-2028x1520-both.json",
    "nighttimeoverlay": "overlay1-RPi_HQ-2028x1520-both.json",
    "cameratype": "RPi",
    "cameramodel": "HQ",
    "cameranumber": 0,
    "lastchanged": "2025-01-03 00:00:26"
}

Alex-developer, 2025-01-03:
If you look in the log you will see

`
2025-01-03T01:11:13.420245+01:00 allsky allsky[10298]: INFO: Not enabled: Moon Sun Satellites Planets`

To correct this goto the overlay editor and in its options enable the parts you require.

Also make sure you read the overlay documentation on adding variables

Hope that helps

Alex

EricClaeys, 2025-01-03:
@Alex-developer, can this message be added to the Overlay Editor and/or sent to `addMessage.sh` ?

Alex-developer, 2025-01-03:
> @Alex-developer, can this message be added to the Overlay Editor and/or sent to `addMessage.sh` ?

No as its not an error its just information thats output when the debug level is 4, the user may now want these turned on

Alex

newgecko, 2025-01-04:
Hi Alex, thanks for the help ... I had interpreted the logs in the same way. I simply forgot to tick the appropriate box because I overlooked the checkbox.
Thanks again. 

Greedings Nico

newgecko, 2025-01-04:
Hi Alex, thanks for the help ... I had interpreted the logs in the same way. I simply forgot to tick the appropriate box because I overlooked the checkbox.
Thanks again. 

Greedings

closed 2025-01-04 by newgecko

Empire4191, 2025-02-08:
Hello,
Unfortunately I also have problems displaying the data from the moon.
The sun is displayed. But not the moon and not the ISS.

[https://www.astro-fotografie.at/moon1.jpg](url)

[https://www.astro-fotografie.at/moon2.jpg](url)

EricClaeys, 2025-02-08:
@Empire4191,
Does `grep ERROR /var/log/allsky.log` show any error messages?

Empire4191, 2025-02-10:
> [@Empire4191](https://github.com/Empire4191), Does `grep ERROR /var/log/allsky.log` show any error messages?

2025-02-10T20:52:07.552262+00:00 allsky allsky[1665]: ERROR: Unable to download de421.bsp: cannot download https://ssd.jpl.nasa.gov/ftp/eph/planets/bsp/de421.bsp because <urlopen error [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: self-signed certificate in certificate chain (_ssl.c:992)>
2025-02-10T20:52:07.552744+00:00 allsky allsky[1665]: ERROR: Cannot use format ':.1f' on Number variables like ${DISKFREE} (value=105.67G).
2025-02-10T20:52:23.058533+00:00 allsky allsky[901]: STARTING EXPOSURE at: 2025-02-10 20:52:23   @ 11.3 sec
2025-02-10T20:52:36.504877+00:00 allsky allsky[901]:   > Saving NIGHT image 'image-20250210205223.jpg'
2025-02-10T20:52:39.734080+00:00 allsky allsky[1853]: ERROR: Unable to download de421.bsp: cannot download https://ssd.jpl.nasa.gov/ftp/eph/planets/bsp/de421.bsp because <urlopen error [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: self-signed certificate in certificate chain (_ssl.c:992)>
2025-02-10T20:52:50.686237+00:00 allsky allsky[901]:      ***** Stopping Allsky *****
2025-02-10T20:52:50.876349+00:00 allsky allsky[703]:      ***** AllSky Stopped *****
2025-02-10T20:53:09.992681+00:00 allsky allsky[713]:      ***** Starting AllSky *****
2025-02-10T20:53:15.472704+00:00 allsky allsky[906]: *****************************************************
2025-02-10T20:53:15.472919+00:00 allsky allsky[906]: *** Allsky Camera Software Version v2024.12.06_01 ***
2025-02-10T20:53:15.472980+00:00 allsky allsky[906]: *****************************************************
2025-02-10T20:53:15.473021+00:00 allsky allsky[906]: Capture images of the sky with a Raspberry Pi and  an RPi camera
2025-02-10T20:53:15.473062+00:00 allsky allsky[906]: Author: Thomas Jacquin - <jacquin.thomas@gmail.com>
2025-02-10T20:53:15.473103+00:00 allsky allsky[906]: Contributors:
2025-02-10T20:53:15.476681+00:00 allsky allsky[906]:  -Knut Olav Klo
2025-02-10T20:53:15.476769+00:00 allsky allsky[906]:  -Daniel Johnsen
2025-02-10T20:53:15.482208+00:00 allsky allsky[906]:  -Robert Wagner
2025-02-10T20:53:15.482342+00:00 allsky allsky[906]:  -Michael J. Kidd - <linuxkidd@gmail.com>
2025-02-10T20:53:15.482383+00:00 allsky allsky[906]:  -Rob Musquetier
2025-02-10T20:53:15.482456+00:00 allsky allsky[906]:  -Eric Claeys
2025-02-10T20:53:15.482504+00:00 allsky allsky[906]:  -Andreas Lindinger
2025-02-10T20:53:15.482683+00:00 allsky allsky[906]:  Camera Information:
2025-02-10T20:53:15.482729+00:00 allsky allsky[906]:   - Type: RPi
2025-02-10T20:53:15.482767+00:00 allsky allsky[906]:   - Model: HQ (HQ)
2025-02-10T20:53:15.482806+00:00 allsky allsky[906]:   - Serial Number: [not supported]
2025-02-10T20:53:15.482843+00:00 allsky allsky[906]:   - Native Resolution: 4056x3040
2025-02-10T20:53:15.482881+00:00 allsky allsky[906]:   - Pixel Size: 1.55 microns
2025-02-10T20:53:15.482919+00:00 allsky allsky[906]:   - Supported Bins: 1 2
2025-02-10T20:53:15.482969+00:00 allsky allsky[906]:   - Color camera, bayer pattern: RG
2025-02-10T20:53:15.483008+00:00 allsky allsky[906]:   - Sensor temperature: 0.0 C
2025-02-10T20:53:15.483047+00:00 allsky allsky[906]:   - Bit depth: 12
2025-02-10T20:53:15.483084+00:00 allsky allsky[906]:   - Auto focus supported: No
2025-02-10T20:53:15.483123+00:00 allsky allsky[906]: Settings:
2025-02-10T20:53:15.483161+00:00 allsky allsky[906]:    Command: rpicam-still
2025-02-10T20:53:15.483216+00:00 allsky allsky[906]:    Image Type: RGB24 (1)
2025-02-10T20:53:15.483254+00:00 allsky allsky[906]:    Resolution (before any binning): 4056x3040
2025-02-10T20:53:15.483292+00:00 allsky allsky[906]:    Configuration file: /home/admin/allsky/tmp/capture_args.txt
2025-02-10T20:53:15.483330+00:00 allsky allsky[906]:    Quality: 95
2025-02-10T20:53:15.483368+00:00 allsky allsky[906]:    Daytime capture: Yes
2025-02-10T20:53:15.483580+00:00 allsky allsky[906]:    Daytime save: No
2025-02-10T20:53:15.483637+00:00 allsky allsky[906]:    Nighttime capture: Yes
2025-02-10T20:53:15.483677+00:00 allsky allsky[906]:    Nighttime save: Yes
2025-02-10T20:53:15.483717+00:00 allsky allsky[906]:    Exposure (day):   500.00 ms (0.50 sec), Auto: Yes, Max Auto-Exposure: 30.0 sec
2025-02-10T20:53:15.483756+00:00 allsky allsky[906]:    Exposure (night):        10.0 sec, Auto: Yes, Max Auto-Exposure: 30.0 sec
2025-02-10T20:53:15.483795+00:00 allsky allsky[906]:    Gain (day):       1, Auto: Yes, Max Auto-Gain: 16
2025-02-10T20:53:15.483846+00:00 allsky allsky[906]:    Gain (night):     8, Auto: Yes, Max Auto-Gain: 16
2025-02-10T20:53:15.483885+00:00 allsky allsky[906]:    Target Mean Value (day):       0.500
2025-02-10T20:53:15.483923+00:00 allsky allsky[906]:    Target Mean Value (night):     0.200
2025-02-10T20:53:15.484078+00:00 allsky allsky[906]:    Target Mean Threshold (day):   0.100
2025-02-10T20:53:15.493414+00:00 allsky allsky[906]:    Target Mean Threshold (night): 0.100
2025-02-10T20:53:15.493487+00:00 allsky allsky[906]:       p0: 5.000
2025-02-10T20:53:15.493527+00:00 allsky allsky[906]:       p1: 20.000
2025-02-10T20:53:15.493565+00:00 allsky allsky[906]:       p2: 45.000
2025-02-10T20:53:15.493604+00:00 allsky allsky[906]:    Binning (day):   1
2025-02-10T20:53:15.493642+00:00 allsky allsky[906]:    Binning (night): 1
2025-02-10T20:53:15.493680+00:00 allsky allsky[906]:    White Balance (day)   Red: 2.50, Blue: 2, Auto:  No
[... cut ...]
2025-02-10T20:54:47.801048+00:00 allsky allsky[906]:   > Saving NIGHT image 'image-20250210205434.jpg'
2025-02-10T20:54:50.991496+00:00 allsky allsky[1158]: ERROR: Unable to download de421.bsp: cannot download https://ssd.jpl.nasa.gov/ftp/eph/planets/bsp/de421.bsp because <urlopen error [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: self-signed certificate in certificate chain (_ssl.c:992)>
2025-02-10T20:55:06.592500+00:00 allsky allsky[906]: STARTING EXPOSURE at: 2025-02-10 20:55:06   @ 11.3 sec

EricClaeys, 2025-02-11:
@Empire4191,

This line in the log file is key:
~~~
ERROR: Unable to download de421.bsp: ...
~~~

This is a known bug, see [Issue #4324](https://github.com/AllskyTeam/allsky/issues/4324) for a workaround.