AllskyTeam/allsky #4164: Ensure fullscreen live view image fits the screen without scrolling

Alex-developer, 2025-01-03:
### Discussed in https://github.com/AllskyTeam/allsky/discussions/4155

<div type='discussions-op-text'>

<sup>Originally posted by **Oxize** January  1, 2025</sup>
Is there a possible to change the current fullscreen function to fit in screen fullscreen instead of sliders?

Its not really a fullscreen picuture. It picks only 3/4 of the picture in fullscreen.

If i click on this Live View screen to go fullscreen. 
[image]

Ìt only picks like 3/4 of the picuture. Not the whole picture. Can this be changed?
[image]

Still missing the lowest part:
[image]

I am Lastest version Chrome Browser. Its the same with Firefox.

</div>

magore, 2025-05-28:
This is a really big problem on mobile devices. I use the rpi-hq camera and the view starts out with a tiny corner of the display zoomed in.also  I can not zoom out enough to fill the screen.

EricClaeys, 2025-05-28:
@magore,
Can you attach a screenshot of what you are talking about?

magore, 2025-05-29:
This is the default zoom followed by Max zoom out. Note also the menu structure on the upper left in the last image is tiny

[image]
[image]

magore, 2025-05-29:
This is a link to am image I did last year during an Arora 
https://photos.app.goo.gl/2aqjPCBNXWCygrjdA

magore, 2025-05-30:
I tried Firefox. It works. I then force stopped Chrome , deleted my cache and restarted it. Now it looks fine. So some kind of cached files were messing it up. FYI I was running Android 15 with chrome.
Thanks
Here is my result after doing those steps above.
[image]

EricClaeys, 2025-05-30:
@magore,
Glad it working now.
Eric

NiteRide, 2025-07-06:
@Oxize - I found that if I need/want to see the whole image in the web UI I just resize the browser window width to be less wide and the image scales to fit the width of the window.  Too bad it's cloudy and rainy or the below might be more interesting to look at.....

[image]

Oxize, 2025-07-06:
That doesnt work for me. And that also depending on what monitor and resolution they use. We just need a fit to screen mechanic and problem is solved.

Alex-developer, 2025-07-07:
This may require an upgrade to some underlying frameworks we use. I'll add this to the next major release list

Oxize, 2025-07-07:
Thanks alex.

Oxize, 2025-08-30:
And luck with this?

Alex-developer, 2025-08-30:
Yes this will come in the next major release, hopefully due soon

Alex-developer, 2026-10-02:
Closing as done

closed 2026-10-02 by Alex-developer