AllskyTeam/allsky #4171: [BUG] Upload path for directory name is "keograms" instead of "keogram"

adrianwells, 2025-01-04:
### Camera model

RPi HQ, sensor imx477

### Operating System

Bookworm

### Allsky version

v2024.12.06

### Pi and amount of memory (512 MB, 1, 2, 4, or 8 GB)

RPi 4 Model B Rev 1.5, 4 GB

### Bug Description

Thanks for this project, I believe this is a bug after upgrading from v2023.05.01_05 to v2024.12.06, found the Keogram image files were no longer being uploaded (via SFTP). It appears the upload path is now expected to be plural "keograms" instead of "keogram". After updating the SFTP directory destination to "keograms" this seems to be working.

Example:
```
 ~/allsky/scripts/generateForDay.sh --upload 20250102

Commands used are in: /home/supa/allsky/tmp/lftp_cmds/Keogram.txt
Upload output from '/home/supa/allsky/images/20250102/keogram/keogram-20250102.jpg:
   *** upload.sh: ERROR, RET=1:
FILE_TO_UPLOAD='/home/supa/allsky/images/20250102/keogram/keogram-20250102.jpg'
REMOTE_HOST='192.168.207.10'
DIRECTORY='/allsky/keograms/'
TEMP_NAME='Keogram-13836'
DESTINATION_NAME='keogram-20250102.jpg'

cd: Access failed: No such file (/allsky/keograms)
```

### Log and configuration files (ATTACH files, do NOT copy/paste them)

[settings.json.txt](https://github.com/user-attachments/files/18307011/settings.json.txt)

EricClaeys, 2025-01-04:
@adrianwells,
The directory name on the Website has always been plural - `keograms`.
The directory name where the daily keogram file is stored on the Pi has always been singular - `~/allsky/ images/ YYYYMMDD/keogram`.

Have you run `remoteWebsiteInstall.sh` ?  If not, I suspect in v2023.05.01_05 you had `KEOGRAM_DIR=keogram`.

adrianwells, 2025-01-04:
@EricClaeys thanks for the reply. I have not yet run `remoteWebsiteInstall.sh` since the upgrade to v2024.12.06. I do not recall adjusting the `KEOGRAM_DIR=` setting but maybe that was the issue in which case I apologize. What caught my attention was that this stopped working after the upgrade (maybe setting was not carried over?).

EricClaeys, 2025-01-04:
@adrianwells, please run `grep KEOGRAM_DIR ~/allsky-OLD/config/ftp-settings.sh`.

Because the new release no longer allows users to change the keogram, startrails, or timelapse directories, the KEOGRAM_DIR, STARTRAILS_DIR, and VIDEOS_DIR settings do not apply and hence aren't carried over.  The installation script should probably check if the user had those set to a "non-standard" location and warn the user.  Not sure how easy it would be to check though.  Maybe instead the installation should disable the remote Website until the user's upgraded it.

I suggest changing the name on your server from `keogram` to `keograms` (which will fix the upload problem), then run `remoteWebsiteInstall.sh`.

EricClaeys, 2025-01-04:
Thanks @adrianwells.  That makes sense.
I am closing this Issue since it's not a bug and I don't see an easy, foolproof way to check if the user had a non-standard directory.

If you have any problems upgrading your Website please add a new Discussion item.
Thanks - Eric

closed 2025-01-04 by EricClaeys

adrianwells, 2025-01-04:
@EricClaeys OK. I edited the title to reflect that this is an issue with the non standard directory name not working after upgrade to v2024.12.06. Thanks for the help and review. It does seem like it may impact other folks upgrading but hopefully they will find this as way to resolve the issue for the time being.