AllskyTeam/allsky #4181: [BUG] Cannot do exposure longer than 11.8s (before updating, it's was usually 60s)

mplu, 2025-01-06:
### Camera model

Picam NoIR v2

### Operating System

Bookworm

### Allsky version

v2024.12.06

### Pi and amount of memory (512 MB, 1, 2, 4, or 8 GB)

1gb

### Bug Description

[settings.json](https://github.com/user-attachments/files/18317510/settings.json)
After updating the allsky application (fresh install os RPi OS and allsky software), i'm not able to do picture longer than 11.8s. Before the update, I used to do exposure of 60s, but it's not working anymore. What point of my configuration is wrong ?

aaronwmorris, 2025-01-07:
If it is the IMX219 module, you should not have been able to expose for more than 11.76s.  This is the documented maximum exposure for the module.

EricClaeys, 2025-01-07:
@mplu,
As @aaronwmorris said, your `Waveshare imx219-d160` camera has a maximum exposure of 11.767.556,0 us per the "rpicam-still" command, which is what Allsky uses to actually take pictures.

Prior versions of Pi OS and of the libcamera software didn't enforce this limit but do now.  You can try contacting Waveshare about having the official limit increased, or add an Issue on the [rpicam-apps GitHub page](https://github.com/raspberrypi/rpicam-apps).  Beware that they may tell you they use whatever Waveshare gives them.

There's nothing Allsky can do about this.

mplu, 2025-01-07:
Indeed, it's the IXM219. So I guess I've got to deal with that... Do a roll back to previous version of RPi OS could be a solution ?

EricClaeys, 2025-01-07:
I think it's the version of libcamera-apps you would need to roll back, plus you would need to stay on the v2023.05.01 release of Allsky since the newest release enforces the same limits.

Bookworm came with the newer libcamera-apps so you would want to try Bullseye, but I don't know how you ensure you never get the newest version.

Personally, I think it would be fighting an uphill battle.  I suggest contacting Waveshare.

mplu, 2025-01-11:
Or change the camera... Is there a resource that summarize the pros, cons, benefits and limitation of various camera that could be used ?

EricClaeys, 2025-01-12:
@mplu,
I didn't want to be the one that said that, but yes, changing cameras is an option.

You can look at `~/allsk/config_repo/RPi_cameraInfo.txt.repo` for a list of all the RPi cameras Allsky supports and what there max exposures are.
I am not aware of anything that sumarizes the cameras.  The RPi HQ camera is by far the most popular and is the one I recommend.  It's significantly cheaper than the cheapest ZWO camera, the resolution is very good, and the quality is good.  If you ignore cost, I don't know why anyone would purchase any other RPi camera.  Auto focus sounds great but with a fixed camera that only looks at the sky, it really only helps to set the initial focus.  Using auto focus at night doesn't really work since it adds a long delay to every picture.

EricClaeys, 2025-01-24:
@mplu,
Is it ok to close this Issue since there's nothing Allsky can do about it?

closed 2025-01-24 by mplu