AllskyTeam/allsky #4192: [FEATURE REQUEST] Generate and upload thumbnail of image.jpg

Hans5546, 2025-01-08:
### If this is a new feature, describe what you would like.

Hello!
I'm using allsky on a RPI Zero 2 W in Sweden - everything works fine, thanks for this excelent software! Keograms, startrails, videos - everything works as expected.

I like to automatically generate a thumbnail of the actual image.jpg which is uploaded every minute.
The original image is really big and I want to use a thumbnail on a different application (Blink), so loading doesn't take so long.

Is there a possibility for an actual thumbnail image?

Many thanks in advance.

EricClaeys, 2025-01-08:
@Hans5546,
Not sure I totally understand what you would like to do.  In the second paragraph it sounds like you only want to upload the thumbnail, but in the 3rd paragraph it sounds like you want to upload both the thumbnail and the image.

By default a thumbnail is generated for each image and stored in `~/allsky/images/YYYYMMDD/thumbnails`.  The thumbnails are only 100 pixels wide so not useful to look at. You can resize images before uploading them, so that might be the way to go if you plan to look at the images on the other machine.

Hans5546, 2025-01-09:
Hi!
Thanks for your answer. Yes, you are right, my requierements are not clearly enough.

I like to upload both: the thumbnail and the image. The thumbnail should be like 1024 pixels wide, 100 pixels wide it a little bit to small.

Do you mean the option "Resize Uploaded Images Width"? I found it in your documetation. Will the software then use the full resolution for timelapse and so and only resize the uploaded image?
This must be exactly what I was looking for...

By the way: Does it make sense to use full resolution of the RPI HQ cam to generate images, keograms and startrails? For the timelapse it had to reduce resolution of the video to 1080, since the RPI zero 2 won't be able to generate higher resolution videos...
Maybe my idea of "full res" is totally nonsens and resizing in the post capturing settings will be the best solution?

EricClaeys, 2025-01-09:
The startrails, keogram, and timelapse are created from the images in ~/allsky/images/YYYYMMDD.  Unless you resized those images, they are full resolution.

For high-resolution cameras like the HQ, I suggest people resize the timelapse.  It often fails to create a full-resolution version, plus, almost no one has a monitor with that high of a resolution, so it's really a waste.  I use half width and half height.

The `Resize Uploaded ...` settings make a copy of the image, resize it, upload it, then delete the copy.  The resized version is not saved on the Pi and it doesn't impact anything else.

Allsky does not currently support uploading two images  to a server.  You could probably create a module that made a copy of the image, resized it to 1024 px, then uploaded it, but you would need to know or bash.

Are you wanting to upload the full image and the 1024 image to the same server, or different servers?

EricClaeys, 2025-01-24:
@Hans5546, any update?

Hans5546, 2025-01-24:
Oh sorry, I forgot to answer...

I changed the following setting in the config.sh:

# Resize images before cropping, stretching, and saving.
IMG_RESIZE="true"
IMG_WIDTH=2028
IMG_HEIGHT=1520

This is doing well and I don't see any difference between this settings and "full res" of the RPI HQ camera.

So (actually) I don't need the option for a thumbnail and the full size image any more :-)

I was afraid, that reducing the resolution would result in bad pictures, this is nonsens as I found out.

EricClaeys, 2025-01-24:
I'm glad you now have what you want.

FYI, you may want to upgrade to the newest Allsky release, which no longer uses `config.sh` or `ftp-settings.sh` - all settings are now in the WebUI.  It also has better troubleshooting and new features.

We will be releasing an update on February 1st which will make using the Overlay Editor easier.

EricClaeys, 2025-01-24:
Closing Issue; user found another way to implement.

closed 2025-01-24 by EricClaeys