AllskyTeam/allsky #4193: [BUG] keeps crashing/freezing after timelapse each morning.

clambert24, 2025-01-08:
### Camera model

rpi hd

### Operating System

Buster

### Allsky version

latest version

### Pi and amount of memory (512 MB, 1, 2, 4, or 8 GB)

pi4b

### Bug Description

every day after its completed its timelapse the video feed crashes/freezes and i have to reboot the system to get the live feed back again and for it to start recording again.  i did notice when i ran cat /var/log/allsky.log | less towards the end it was producing ALLOT of extra info i have never seen before, in this case (added extra data field)

clambert24, 2025-01-08:
[image]

EricClaeys, 2025-01-09:
All the extra log entries are because you have `Debug Level` set to 4.  Try using 3.

We can't do much without the log file.  If you have the Pi desktop you can use it's browser and drag and drop the file.
You can also run `allsky-config` and pick the option to easily share files with a PC. You can then copy `/var/log/allsky.log` to your PC or mac and upload it.

clambert24, 2025-01-09:
Ok that makes sense. 
Yes i do have the pi desktop. Where would I find the file?

EricClaeys, 2025-01-09:
Which file?

The Pi browser is on the Pi's Taskbar - on mine it's a blue globe.

The "log" file is `/var/log/allsky.log`.

clambert24, 2025-01-10:
im so sorry, im so confused. 
i clicked on the Pi browser, then what? i dont see anything about the file /var/log/allsky.log????

EricClaeys, 2025-01-11:
On the Pi desktop, open the Pi browser and go to the Allsky GitHub page and log in, just like you do on your PC.  Click on the `Issues` link then open your Issue.
Start a reply but don't type anything.

Click on the File Explorer icon on the Pi desktop (I think it looks like a folder, similar to Windows).  In that window, go to `/var/log`.  You should see several files, including `allsky.log`.
Drag the `allsky.log` icon onto your GitHub reply, then submit your reply.

clambert24, 2025-01-12:
I'm sorry, but I literally can't find /var/log anywhere. 
The closest I can find anything to do with log is in /allsky/config/logs. 
[image]

EricClaeys, 2025-01-12:
@clambert24,
I strongly recommend reading the `Basics -> Linux` documentation page.  The few minutes it takes will save you a lot of time and frustration going forward.  And it explains pathnames which is a confusing concept for many new users.

Eric

clambert24, 2025-01-13:
ok. so think i have managed to work out how to do the log etc. does this help?

[settings.json.txt](https://github.com/user-attachments/files/18400896/settings.json.txt)
[allsky.log](https://github.com/user-attachments/files/18400898/allsky.log)

EricClaeys, 2025-01-13:
@clambert24, perfect.  Thanks for attaching the files.  Did the Linux Basics page help?

Did the Pi crash or freeze after the last entry in the log file "Posting camera details to Allsky map" ?
When you say "freeze", what freezes?  Allsky, your terminal window, the Pi desktop, etc.?

FYI, Support for Buster is going away soon, so I suggest upgrading to Bookworm 64-bit when you can.  I still have to upgrade one of my Pi's.

clambert24, 2025-01-13:
yes it did help, but was still a little confusing as i do like a step by step guide.

the picture freezes in the morning after the timelapse/sunrise has happened.  i can still login remotely afterwards etc, but if i leave it, it doesn't start recording again at night etc, so i have to go into settings and reboot pi for it to continue doing what it should.

FYI. i am running on Bookworm 64-bit as iv just checked.

clambert24, 2025-01-13:
ok thank you. 
yes its exactly the same as i had left it most of the day "frozen" until i got home from work and thats all the info it provided. 

ok so i shall give that a go and i will see tomorrow what happens, its just very confusing thats all because it was working fine with no problems.

EricClaeys, 2025-01-13:
I don't think this is an Allsky bug.  I'm pretty sure other people have had this problem and it was caused by Linux killing Allsky because it was using "too much" CPU.
We will see tomorrow.

aaronwmorris, 2025-01-14:
@clambert24 Assuming you are using a Raspi HQ camera.  The reason your system is hanging is most likely due to running out of memory.  If you are trying to generate a timelapse at the native resolution of 4056x3040, ffmpeg will require ~2.5GB of memory (more than your 2GB available).

The amount of memory needed by ffmpeg is relative to the output resolution of the video (not the input images).  You could try adding `--vf scale=-2:1080` or `--vf scale=-2:2160` to your ffmpeg command to reduce the vertical height of your videos to 1080p or 2160p.  These would only require 0.6GB and 1.7GB of memory respectively.

I would also suggest setting up the watchdog function on your Pi.  This will automatically reboot the system if it hangs.

clambert24, 2025-01-14:
Hi. Ok that is good information thank you. 
Yes i am using that camera. 
I shall try adding one of those commands to see if that helps. 
Where would I find info for setting up the watchdog function?

aaronwmorris, 2025-01-14:
I could not find anything using a quick search of the allsky docs.  I have a pretty easy guide in the indi-allsky wiki:

https://github.com/aaronwmorris/indi-allsky/wiki/Raspberry-Pi-Watchdog

rodobie, 2025-01-16:
I had a similar timelapse issue which I root caused back to the USB C power supply.

Was using a spare USB plug which I had for charging mobiles, to run my PI 5 8Gb  and it was fine for everything except when the PI was on at full load for extended duration. It could not supply the necessary power and would  cause a crash at a random point during the timelapse creation.

Bought the  recommended Raspberry PI5 27W USB-C power supply and have never had a timelapse creation crash since installing it.

Hope this helps you out.

clambert24, 2025-01-16:
> [@clambert24](https://github.com/clambert24) Assuming you are using a Raspi HQ camera. The reason your system is hanging is most likely due to running out of memory. If you are trying to generate a [...]

Ok, so I applied the setting --vf scale=-2:2160 and now it's not generating a timelapse 😕.

[image]

EricClaeys, 2025-01-17:
@clambert24,
You should NEVER change an Allsky script and almost never manually change a configuration file.  Configuration changes are done via the WebUI.  **Please undo whatever you did.**

To change the size of the timelapse video update the Timelapse `Width` and `Height` settings in the WebUI.
For my RPi HQ camera I set them to half the width and height of the sensor (4056x3040) since that keeps the same aspect ratio.

Going forward, please remember that the Allsky Documentation is your friend.  It describes every setting and explains what most things are and how to use them, like timelapse.  There's also a ton of troubleshooting info in the Documentation.
If you find something that's confusing or missing, please let us know.

I'm curious what the "Images" screenshot is supposed to tell us?
The `Explanations -> Timelapse` documentation page tells you when timelapses are automatically created and how to manually create them.

clambert24, 2025-01-18:
> [@clambert24](https://github.com/clambert24), You should NEVER change an Allsky script and almost never manually change a configuration file. Configuration changes are done via the WebUI. **Please [...]

Hi.

ok i have undone what the previous person said and changed the width and height settings as you mentioned. 

the above picture was showing that the timelapse wasnt generated when it should of been.

EricClaeys, 2025-01-19:
> the above picture was showing that the timelapse wasnt generated when it should of been.

Are you referring to the screenshot of the "Images" page?
Recall that the timelapse, keogram, and startrails are created at the end of night which is around 6 am **the next day.**
The top row of the "Images" page is for the current day, and because those files aren't created until the next day, you'll always see "none" for them.  Because some images have been taken on the current day, you'll always see the image icon for the current day.
So, the screenshot is correct.

EricClaeys, 2025-01-19:
@clambert24,
is everything working now?

clambert24, 2025-01-19:
Yes it recorded last night and saved it. 
I shall see if it records again tonight. I did notice that the video once again had fuzziness due to low cloud but I had removed that problem by putting in another setting (can't remember what that was now)

EricClaeys, 2025-01-20:
@clambert24,
Thanks for the update.
I think Allsky didn't "freeze" after the timelapse was created; it was forcefully killed, so couldn't post the "Allsky is not Running" message which meant the last image never got updated and it looked like it froze.

I'm pretty confident that reducing the resolution of the timelapse will fix that problem.  And it's highly likely the `-level 3.1` will fix (or at least improve) the fuzzyness.

If everything seems ok, please close the issue. 
Thanks

clambert24, 2025-01-20:
hi. ok so i changed the resolution, but it still froze. i have included my log again for you and hope this helps?

Chris. 

[allsky.log](https://github.com/user-attachments/files/18482274/allsky.log)

EricClaeys, 2025-01-24:
@clambert24,
would you please attach your `~/allsky/config/settings.json` file?  You'll need to append `.txt` to its name.

What power supply are you using?  Someone else whose Pi crashed after timelapse was using a power supply that didn't supply enough power.

clambert24, 2025-01-25:
HI. ok so i have hopefully included the correct file you need.

Im using the official raspi pi psu (27w and not even a year old).
its just odd because it was all working fine with no problems, and only really started since updating the allsky software to the latest version.

[settings.json.txt](https://github.com/user-attachments/files/18546703/settings.json.txt)

clambert24, 2025-01-25:
because you said yourself that was the size you had it at in a previous message.

(To change the size of the timelapse video update the Timelapse Width and Height settings in the WebUI.
For my RPi HQ camera I set them to half the width and height of the sensor (4056x3040) since that keeps the same aspect ratio.)

EricClaeys, 2025-01-25:
Ah, I  wasn't clear.  Sorry.
I meant the size of the sensor is (4056x3040) and I use half of that (2028 x 1520).

clambert24, 2025-01-25:
Ah ok, thats not a problem. i shall re size it now and hopefully that cures my problem. thank you.

EricClaeys, 2025-01-25:
Please keep me informed.  I'm 90% certain it'll fix the problem.
Until a few months ago, the Pi wouldn't crash when creating a timelapse; instead, the timelapse program would be aborted an a message added to the log file.  This was easy to determine what happened.
Now, a few people have reported that the Pi crashes.  Something in Linux must have changed.

clambert24, 2025-01-30:
Ok. So I think I finally got success and it working. So now currently it has been recording for about 4 days without it crashing/locking up. 
Looks like reducing timelapse resolution size worked. Can I just ask, does reducing that cause it to loose quality in the recording? 
But yes so far happy with the results. Thank you very much.

EricClaeys, 2025-01-31:
@clambert24,
Technically, reducing the resolution reduces the quality, but unless you looked at the timelapse at 100% (which wouldn't all fit on your monitor), your browser was reducing the resolution to whatever your window size was.  So, it's no worse now than before.
Assuming you have a plastic dome, it probably reduces quality due to distortions more than any decrease in resolution.

I'm closing this Issue since you have things working now.

closed 2025-01-31 by EricClaeys