AllskyTeam/allsky #4197: Startrails not generating correctly

jorksx, 2025-01-12:
### Camera model

ZWO ASI224MC

### Operating System

Bookworm

### Allsky version

v2024.12.06

### Pi and amount of memory (512 MB, 1, 2, 4, or 8 GB)

pi 5

### Bug Description

[image]

Startrails image is not showing me whole picture, seems like only 1 frame.
90% of the night, sky was clear, full of starts like in following picture: 
[image]

But startrails image generated by allsky seems to be wrong.
Is this user problem or am I missing someting?

EricClaeys, 2025-01-12:
@jorksx,
Did you look at the `Startrails` documentation page?  It describes how to configure startrails.

EricClaeys, 2025-01-22:
@jorksx, any update?

jorksx, 2025-01-22:
Nope. I looked at the documentation and tried playing with settings etc...
Raised the brightness threshold to 0.4, but still I get only one image with brightness of 0.28 and nothing else, even tough 90% of night pictures are around 0.30

EricClaeys, 2025-01-28:
@jorksx, can you zip a few dozen nighttime images and attach to your reply so I can try?
If GitHub doesn't allow that big of an attachment, can you copy them elsewhere?

EricClaeys, 2025-02-19:
@jorksx, are you able to attach a few dozen images?

EricClaeys, 2025-05-10:
Closing Issue.  No response from user in 3 months.  Other users are able to get startrails working by adjusting the Threshold value.

closed 2025-05-10 by EricClaeys