AllskyTeam/allsky #4200: [BUG] installation fails on installing numpy

notterme, 2025-01-13:
### Camera model

RPi HQ IMX477

### Operating System

Bookworm

### Allsky version

v2024.12.06

### Pi and amount of memory (512 MB, 1, 2, 4, or 8 GB)

RPi 4 with 4GB

### Bug Description

Brand new RPi, bookworm, fully apt-get updated and rebooted.

Running install as normal user

Following instructions as per: 
https://htmlpreview.github.io/?https://raw.githubusercontent.com/AllskyTeam/allsky/master/html/documentation/installations/Allsky.html

After clone from github and running ./install the process stops with:

*** ERROR: 
**********
Python_dependencies [numpy>=1.25.0] failed
**********

* The full log file is in /home/map/allsky/config/logs/Python_dependencies.1.log
The end of the file is:
          
          note: This error originates from a subprocess, and is likely not a problem with pip.
        error: metadata-generation-failed

        Encountered error while generating package metadata.

        See above for output.

        note: This is an issue with the package mentioned above, not pip.
        hint: See above for details.

/var/log/allsky.log is 0 Bytes.

Don't know how to fix this. Please help.

[Python_dependencies.1.log](https://github.com/user-attachments/files/18394437/Python_dependencies.1.log)
[install.log](https://github.com/user-attachments/files/18394461/install.log)
[settings.json.txt](https://github.com/user-attachments/files/18394480/settings.json.txt)

EricClaeys, 2025-01-13:
@Alex-developer, any ideas?

notterme, 2025-01-19:
I tried to recreate the issue and was 'successful' (and it failed the installation) once, but then I had two successful installs, every time dd-ing the RPi-OS image to disk etc..

I'm starting to believe that I may not have done a full apt-get upgrade before commencing the install.

Probably user error.

Feel free to close the issue.

EricClaeys, 2025-01-19:
Closing Issue per user request.

EricClaeys, 2025-04-25:
@morrowwm, not poor form at all.  I've done it.
It's had to know which Discussion and Issues we should close and which should stay open.
If one is closed, it's harder for people to find so the risk of duplicate reports goes way up.  But if we leave everything open there would be so many items it'd be almost impossible to find anything.

The newest release of Allsky has a very helpful "Getting Support" WebUI page that produces a .zip file with a lot of information, including the log file.  So far it has always produced the information we needed to solve the problem.

If you still have the Allsky that didn't install correctly, please attach it's `config/log/install.log` file.

morrowwm, 2025-04-28:
I wiped it out, @EricClaeys, sorry. The error was essentially the same as @notterme experienced. The OS had numpy 1.24._something_ 

Strangely, when I redid the flash of `2024-11-19-raspios-bookworm-arm64-full` on the SD card, configured debian and rebooted, and then installed allsky, it all went OK. 

So maybe I used a 32 bit OS? Or some other glitch with the OS