AllskyTeam/allsky #4226: [BUG]

james-dmu, 2025-01-21:
### Camera model

Raspberry pi HQ Camera/Arducam Owlsight 64MP camera

### Operating System

Bookworm

### Allsky version

v2024.12.06

### Pi and amount of memory (512 MB, 1, 2, 4, or 8 GB)

512Mbyte on RPi Zero 2 W (HQ camera) and 2GB on Raspberry Pi 4 (Owlsight cameras)

### Bug Description

It appears that the image resize settings (for height and width) in the Web GUI fail to apply and reset to 0. I have tested this on 3 systems and it doesn't appear to matter what numbers you enter in the Web GUI the setting does not save and reverts to 0.

I have look at the setting file that should change and that stays at 0 also. As a test I have manually changed the setting file and then the image resize works and the Web GUI reflects the setting correctly however it still cannot be changed in the Web GUI. As a result it appears to just be an issue with the Web GUI not writing the change to the setting file.

There's nothing logged that I am aware of.

EricClaeys, 2025-01-21:
Thanks for reporting it.

The fix will be in the next release on February 1st.  If you can't wait till then, let me know and I should be able to give you the file that's updated.

Eric

james-dmu, 2025-01-22:
Hi Eric,

If you already have the fixed file and it's no hassle I can give it a try but otherwise I can use the workaround of setting it manually in the settings file in the meantime.

On a somewhat unrelated note I just wanted to say thanks for adding the support for the Arducam Owlsight 64MP cameras. The installation detected it and it seems to be working fine, :)

EricClaeys, 2025-01-22:
@james-dmu,
the fix was more extensive than I thought so there are 6 files that changed.
Since you have a workaround, let's wait for the release on February 1.  You'll want to upgrade when it's out.

I'm glad the Arducam is working for you.

FYI, this Issue will automatically close when the new release is out.

closed 2025-02-08 by EricClaeys