AllskyTeam/allsky #4243: [BUG] WebGUI credentials only work using .local, IP or just hostname. Not with fully qualified network name

james-dmu, 2025-01-26:
### Camera model

N/A to bug report

### Operating System

Bookworm

### Allsky version

v2024.12.06

### Pi and amount of memory (512 MB, 1, 2, 4, or 8 GB)

Raspberry Pi 3, Zero 2W and 4

### Bug Description

I'm not sure what the root cause of this issue is and it probably won't affect a lot of people however before upgrading to the latest Allsky (v2024.12.06) I was able to access my camera via it's fully qualified domain name of "orion.lan" however now if I use .lan (my networks domain) in place it will refuse the username/password. It just rejects it and brings the credentials box back up.

If I use just "orion" or "orion.local" or IP it works. This wasn't the case with the previous version that responded correctly if I used the FQDN. The network settings are correct on the Pi and I can ping and SSH to the cameras using the FQDN but just cannot log into the WebGUI that way.

A relatively minor issue admittedly but it seems like using the systems FQDN should work too as it did before. If I disable the username/password requirement then I can access the system via "orion.lan" it just seems that the authentication doesn't like that domain name and is attached to .local.

EricClaeys, 2025-01-26:
@Alex-developer, can you think of anything in v2024.12.06 that would cause this?  I can't. I am not aware of anything in Allsky that knows or cares what URL it's being accessed from.

Alex-developer, 2025-01-26:
This is a weird one. It sounds like the authorization headers are not being set.

What DNS server are you using to for the .lan addresses? I can set something up here and test it out

Alex

james-dmu, 2025-01-27:
Hi Alex,

I'm using a local Pihole server as DHCP and DNS. Machines locally can resolve everything fine just. It's strange that the credentials fail only if using it's full .lan name. I only noticed as I had links setup to go to the WebGUI and they all used the full .lan as sometimes when I am VPN'ed in (not the case for this issue) it's better to use it's FQDN as it's on a different subnet.

Alex-developer, 2025-01-27:
> Hi Alex,
> 
> I'm using a local Pihole server as DHCP and DNS. Machines locally can resolve everything fine just. It's strange that the credentials fail only if using it's full .lan name. I only noticed as I had links setup to go to the WebGUI and they all used the full .lan as sometimes when I am VPN'ed in (not the case for this issue) it's better to use it's FQDN as it's on a different subnet.
> 
> Just to add to this sorry I have 3 installations of Allsky on different RPi's and the issue is the same on all. This wasn't the case on previous versions of Allsky so that's what made me think something may be different.
> 
> Previous installations were on both Bullseye and Bookworm and so I'm not sure it's related to the OS unless the is something with the very latest Bookworm distro as previous installs were an older version of Bookworm.

Thanks

I'll setup pihole and give it a try, probably won't be until the weekend though.

I did setup bind and try it and all worked fine, its possible we changed something in the lighttpd conf that pihole doesnt like.

I'll update when I have tested it

EricClaeys, 2025-01-28:
I would be very surprised if this was an Allsky issue.
Let's check, though.  Restore the prior release by running:
~~~
cd ~/allsky
./install.sh --restore
~~~
Takes about a minute.  When done, restart Allsky and check access via .lan

james-dmu, 2025-01-28:
Hi Eric,

When I tried out the latest version it was on a different SSD so it wasn't an upgrade. Reverting to the older installs on the other SSDs brings back original functionality as in the .lan name works fine with the WebGUI. One thing that is different is that all the older installs were on either Bullseye or an older distro of Bookworm compared to the current latest so maybe the issue is there rather than with the Allsky software.

If you have more important bugs to deal with then I'm not too concerned about this but just thought it was worth mentioning just in case it affects others.

EricClaeys, 2025-02-03:
@james-dmu,
If you still have the `allsky-OLD` directory on one of the machines that isn't working, try to reinstall that version and see if the problem goes away:
~~~
cd /etc/lighttpd
sudo cp lighttpd.conf lighttpd.conf.v2024.12.06
sudo systemctl stop allsky
mv allsky allsky-NEW
mv allsky-OLD allsky
cd allsky
./install.sh       # if asked, tell it to restart the installation from scratch
~~~

That will set up the lighttpd web server using the prior configuration file.

If that does NOT work, then I'm not sure what to say.  It's likely not a problem with the new release.
If that DOES work, then please attach `~/allsky/config_repo/lighttpd.conf.repo` and I'll compare that to the new one to see what's different.

EricClaeys, 2025-02-08:
@james-dmu,
any update?

james-dmu, 2025-02-10:
Hi Eric, Sorry for the delay replying, been quite busy. All the installs I have are new clean installs so I haven't been able to check what you suggested. Don't worry too much about pursuing this if no one else is reporting it. It may not be related directly to the new version.

EricClaeys, 2025-02-10:
@james-dmu,
No one else has reported this problem, and about 200 people have installed this new release, so I assume it's something specific to your environment.  Hence, I'm closing this Issue.

closed 2025-02-10 by EricClaeys