AllskyTeam/allsky #4246: [BUG] libpng error: Write Error

spikked27, 2025-01-27:
### Camera model

Raspberry Pi HQ Camera

### Operating System

Bookworm

### Allsky version

v2024.12.06

### Pi and amount of memory (512 MB, 1, 2, 4, or 8 GB)

Pi 5 8GB

### Bug Description

Fresh install of allsky, camera seems to run fine but after reviewing the images I have run into many corrupt images over the course of the night (see below screenshot), logs show repeating

```
2025-01-25T23:54:49.718061-05:00 allsky allsky[82689]: libpng error: Write Error
2025-01-25T23:54:50.047480-05:00 allsky allsky[82711]: convert-im6.q16: unable to open image `/home/pi/allsky/tmp/image-20250125235436.png': No such file or directory @ error/blob.c/OpenBlob/2924.
2025-01-25T23:54:50.047570-05:00 allsky allsky[82711]: convert-im6.q16: no images defined `/home/pi/allsky/images/20250125/thumbnails/image-20250125235436.png' @ error/convert.c/ConvertImageCommand/3229.
2025-01-25T23:54:50.048312-05:00 allsky allsky[82655]: *** saveImage.sh: WARNING: THUMBNAIL resize failed; continuing.
2025-01-25T23:54:50.049147-05:00 allsky allsky[82712]: cp: cannot stat '/home/pi/allsky/tmp/image-20250125235436.png': No such file or directory
2025-01-25T23:54:50.049232-05:00 allsky allsky[82655]: *** ERROR: saveImage.sh: unable to copy /home/pi/allsky/tmp/image-20250125235436.png ***
2025-01-25T23:54:50.353865-05:00 allsky allsky[82729]: *** upload.sh: ERROR: File to upload '/home/pi/allsky/tmp/image-20250125235436.png' not found!
2025-01-25T23:54:50.355452-05:00 allsky allsky[82731]: mv: cannot stat '/home/pi/allsky/tmp/image-20250125235436.png': No such file or directory
```

And resulting missing images:
[image]

### Log and configuration files (ATTACH files, do NOT copy/paste them)

[allsky.log](https://github.com/user-attachments/files/18559896/allsky.log)

[settings.txt](https://github.com/user-attachments/files/18559920/settings.txt)

EricClaeys, 2025-01-27:
This is what I would expect to see when the disk is bad or there is some other hardware problem.
Is this a new SD card?
A new Pi 5?

It's also possible the disk is physically good but got corrupted.  Try searching for "linux check disk".  It used to be the `fsck` command (File System ChecK).

spikked27, 2025-01-28:
> This is what I would expect to see when the disk is bad or there is some other hardware problem. Is this a new SD card? A new Pi 5?
> 
> It's also possible the disk is physically good but got corrupted. Try searching for "linux check disk". It used to be the `fsck` command (File System ChecK).

I'll have to pull the SD card to do a fsck on it, as far as hardware problems are we talking camera connection?  Actual issue with the raspberry pi (it is a brand new pi)?  Or could it be a power issue?  I don't see a throttling notification in the system page though.

EricClaeys, 2025-01-28:
Not likely a camera connection issue.  Usually those result in errors while taking the image.  Your errors are after the image is taken and saved.
After looking at the errors again, I think it's most likely the disk. The first error, `Write error` is almost always disk-related.
I advise copying the `allsky` directory to a thumb drive in case you need to replace the SD card.

spikked27, 2025-01-28:
> Not likely a camera connection issue. Usually those result in errors while taking the image. Your errors are after the image is taken and saved. After looking at the errors again, I think it's most [...]

I'll definitely pull a copy of everything over winscp and order another large SD card, won't be able to swap until tomorrow night, will report back thanks for the help!

spikked27, 2025-01-31:
Ok was finally able to swap the SD card for a new one yesterday and let it run overnight and still have corrupt images this morning, log attached below but I'm not seeing the same errors and if I click on one of the images that appear corrupt it eventually opens up

[image]

spikked27, 2025-01-31:
[allsky.txt](https://github.com/user-attachments/files/18620116/allsky.txt)

EricClaeys, 2025-02-03:
@spikked27,

The log file you attached doesn't have any errors in it and you said "I'm not seeing the same errors" so does that mean it's working fine now?

spikked27, 2025-02-03:
> [@spikked27](https://github.com/spikked27),
> 
> The log file you attached doesn't have any errors in it and you said "I'm not seeing the same errors" so does that mean it's working fine now?

It appears that the images are in fact saved but the thumbnail is "broken" in the gallery view, confirmed that the thumbnail exists though, it just won't load for some reason

EricClaeys, 2025-02-03:
How many thumbnails are there?
Go to the page again, then take a look in `/var/log/lighttpd/error.log` to see if there are any errors.

Would you please attach one of the "broken" thumbnails?  You'll need to append `.txt` to its filename first.

spikked27, 2025-02-04:
[error.txt](https://github.com/user-attachments/files/18658177/error.txt)

[image-20250130064814.txt](https://github.com/user-attachments/files/18658194/image-20250130064814.txt)

Nothing in the error.log file I don't think, thumbnail attached

EricClaeys, 2025-02-05:
Thanks.  I was able to view the thumbnail on my phone, so it doesn't appear to be broken.  That implies the problem is with the web page that displays the thumbnails.

Try this:
~~~
cd ~/allsky/images
mkdir -p test/thumbnails
~~~
then copy a dozen or so images and their thumbnails from `images/20250131` to `images/test` and `images/test/thumbnails `.
Then go to the `Images` page and view the `test` directory.

EricClaeys, 2025-02-27:
@spikked27,
Have you had a chance to do what I mentioned in my last post?

EricClaeys, 2025-04-06:
Closing - no response from user in 2 months.