AllskyTeam/allsky #4274: [BUG] AS_DEWCONTROLHEATER not displaying correct value in overlay

astik91, 2025-02-01:
### Camera model

RPi HQ

### Operating System

Bookworm

### Allsky version

v2024.12.06

### Pi and amount of memory (512 MB, 1, 2, 4, or 8 GB)

Raspberry Pi 4 Model B Rev 1.5 (8 GB)

### Bug Description

I have the ${DEWCONTROLHEATER} variable in my overlay. The AS_DEWCONTROLHEATER in /home/pi/allsky/config/overlay/extra/allskydew.json says "On", the heater is indeed on, but the value in overlay always says Off. The other values from Dew Heater module are read and displayed correctly.
I didn't use any "invert" setting in the module.

Here's part of periodic log (also added full file)
2025-02-01T15:19:34.327275+01:00 allsky allskperiodic[88122]: INFO: --------------- Running Module allsky_dewheater ---------------
2025-02-01T15:19:34.327323+01:00 allsky allskperiodic[88122]: INFO: Temperature below forced level 15
2025-02-01T15:19:34.327365+01:00 allsky allskperiodic[88122]: INFO: Turning Heater on using pin 15
2025-02-01T15:19:34.327408+01:00 allsky allskperiodic[88122]: INFO: Sensor DHT22 read. Temperature 13.5 Humidity 29.8 Relative Humidity None Dew Point -3.81 Heat Index -14.47 Pressure None Altitude None

### Log and configuration files (ATTACH files, do NOT copy/paste them)

[allskyperiodic.log](https://github.com/user-attachments/files/18629143/allskyperiodic.log)

Alex-developer, 2025-02-02:
Can you post the contents of

`~/allsky/config/overlay/extra/allskydew.json`

and

`~/allsky/config/overlay/config/userfields.json`

Thanks

astik91, 2025-02-03:
Sure, here it is.
What also bothers me, the dewheater module seems to be always turning on the releay, even though the log says "turning heater off". If switch it off manually, allsky turns it on again on next periodic run. Also tried a different pin with the same result.

[allskydew.json](https://github.com/user-attachments/files/18639671/allskydew.json)

[userfields.json](https://github.com/user-attachments/files/18639673/userfields.json)

Alex-developer, 2025-02-03:
> Sure, here it is. What also bothers me, the dewheater module seems to be always turning on the releay, even though the log says "turning heater off". If switch it off manually, allsky turns it on again on next periodic run. Also tried a different pin with the same result.
> 
> EDIT: I realized I have a low level trigger relay module, so I tried the "Invert Relay" option. While allsky is switching heater on and off, there is no change with the relay, it stays on.
> 
> EDIT2: The relay now works as expected after I connnected it to 3V3 + "Invert Relay". But the issue with Heater overlay variable remains.
> 
> [allskydew.json](https://github.com/user-attachments/files/18639671/allskydew.json)
> 
> [userfields.json](https://github.com/user-attachments/files/18639673/userfields.json)

Try changing the type of the DEWCONTROLHEATER variable to text and remove everything from the format field

astik91, 2025-02-04:
Thanks Alex, that fixed it.

closed 2025-02-05 by EricClaeys