AllskyTeam/allsky #4278: Prior Website files being copied when they shouldn't

EricClaeys, 2025-02-02:
### Discussed in https://github.com/AllskyTeam/allsky/discussions/4267

<div type='discussions-op-text'>

<sup>Originally posted by **evripidis** January 31, 2025</sup>
Hello!

My fresh install is with v2024.12.06_01

While everything runs smoothly and I do not accept to bring back any video, files, configs etc, I've noticed that after activating the AllSky Settings (just with the defaults) and going to Editor I do see my previous configuration.json file.

Is there something that I am missing about file's location?</div>

<hr>

### DESCRIPTION:

When `allsky-OLD` exists but the users tell install.sh NOT to use it, the prior Website files are still copied to the new release.

EricClaeys, 2025-02-02:
There lines indicate the problem:
~~~
??? 31 ??? 2025 07:34:47 ?? EET * Local Website files:
??? 31 ??? 2025 07:34:47 ?? EET * timelapse videos: NO PRIOR VERSION
??? 31 ??? 2025 07:34:47 ?? EET * keograms: NO PRIOR VERSION
??? 31 ??? 2025 07:34:47 ?? EET * startrails: NO PRIOR VERSION
??? 31 ??? 2025 07:34:47 ?? EET * 'myFiles' directory: NO PRIOR VERSION
??? 31 ??? 2025 07:34:47 ?? EET * 'myImages' directory: NO PRIOR VERSION
??? 31 ??? 2025 07:34:47 ?? EET * configuration.json (copying and updating for version 3)
??? 31 ??? 2025 07:34:47 ?? EET * configuration.json (copying)
??? 31 ??? 2025 07:34:47 ?? EET * Updating version from 2 to 3.
~~~

Two other problems.
1: the latitude and longitude are looked at twice.
~~~
??? 31 ??? 2025 07:32:58 ?? EET * Latitude is missing, Longitude is missing; prompting.
??? 31 ??? 2025 07:32:59 ?? EET * Verifying pre-determined Latitude and Longitude.
??? 31 ??? 2025 07:33:55 ?? EET * Latitude set to 40.6436N.
??? 31 ??? 2025 07:33:55 ?? EET * Longitude set to 22.9309E.
...
??? 31 ??? 2025 07:34:42 ?? EET * OVERLAY_NAME: nighttimeoverlay = overlay-ZWO.json, TYPE=text
??? 31 ??? 2025 07:34:42 ?? EET * Verifying pre-determined Latitude and Longitude.
??? 31 ??? 2025 07:34:46 ?? EET * Latitude set to 40.6436N.
??? 31 ??? 2025 07:34:47 ?? EET * Longitude set to 22.9309E.
??? 31 ??? 2025 07:34:47 ?? EET * env.json (creating)
~~~

2: When RAM size can't be determined (probably because the user has the "jammy" operating system), a warning message should be displayed.  Because "" is less than 512, the correct amount of swap is determined, however.
~~~
??? 31 ??? 2025 07:31:39 ?? EET * RAM_SIZE=, SUGGESTED_SWAP_SIZE=4608.
~~~

EricClaeys, 2025-02-02:
You are running "jammy" which not a Pi OS so you're likely to have other problems.  I noticed in the installation output that it couldn't get the RAM size.  In the screenshot above it can't determine the Pi Model.  Allsky requires a Pi OS (buster, bullseye, or bookworm) which have the `vcgencmd` command.
The `sudo` message probably isn't due to lack of `vcgencmd` but is probably related to a different operating system.

EricClaeys, 2025-02-03:
@evripidis,

>  to try directly a Pi OS installation to check that

I'm not sure what you mean?  The install.sh script should check the OS?  Or you should install a Pi OS on your mini PC?

evripidis, 2025-02-03:
Format & install Pi OS.

EricClaeys, 2025-02-03:
Will Pi OS work on a non-Pi devices?

evripidis, 2025-02-03:
Yes https://www.raspberrypi.com/software/operating-systems/#raspberry-pi-desktop

EricClaeys, 2025-02-03:
I fixed the problem with the prior Website files being copied to a new release, even if a user said they didn't want allsky-OLD to be used.

EricClaeys, 2025-02-03:
> Yes https://www.raspberrypi.com/software/operating-systems/#raspberry-pi-desktop

Didn't know that.  Is this a virtual machine running Pi OS, or is it installed on the hardware?

evripidis, 2025-02-03:
No virtiual machine, it installs directly on H/W. 

But now I've just noticed that there is only a 32bit version.

evripidis, 2025-02-04:
> You are running "jammy" which not a Pi OS so you're likely to have other problems. I noticed in the installation output that it couldn't get the RAM size. In the screenshot above it can't determine [...]

Yes you are correct about the sudo, it is due to different system.

Despite that, the point is that the project works great with no issues in my mini pc with Lubuntu distro.

It would be great if the project could be able to verify the system (Pi OS or other linux distro) and read RAM and temperature values.

closed 2025-02-08 by EricClaeys

EricClaeys, 2025-02-27:
@evripidis,
would you please run these on your Jammy system:
~~~
free --mebi
grep MemTotal /proc/meminfo
~~~

I'm trying to find out how to determine the amount of RAM on Jammy.

evripidis, 2025-02-27:
> [@evripidis](https://github.com/evripidis), would you please run these on your Jammy system:
> 
> ```
> free --mebi
> grep MemTotal /proc/meminfo
> ```
> 
> I'm trying to find out how to determine the amount of RAM on Jammy.

Yeah sure!

lubuntu@lubuntu-ami:~$ free --mebi
grep MemTotal /proc/meminfo
               total        used        free      shared  buff/cache   available
Mem:            5709        2679         565         229        2988        3029
Swap:            511         255         256
MemTotal:        5846420 kB
lubuntu@lubuntu-ami:~$

EricClaeys, 2025-02-27:
@evripidis, thanks.  The next release will first try `vcgencmd` to get the RAM, and if it's not found, will use "free --mebi", so it should work will Jammy.

I'd like to identify all the other places where non-Pi OS machines don't work.  Do you know any other places?

evripidis, 2025-02-27:
Hmmm only one other think I can recall is that the vcgencmd is not working on a non pi unit.

EricClaeys, 2025-02-27:
ok, thanks.  `vcgencmd` is only used in a few places in Allsky so I should be able to find workarounds in the other places.

evripidis, 2025-02-27:
I think that I was trying to bring my unit's temp in the overlay, that's how I "discovered it". But then I've just let it go and never search it much deeper.