AllskyTeam/allsky #4324: de421.bsp not downloading

EricClaeys, 2025-02-11:
### Discussed in https://github.com/AllskyTeam/allsky/discussions/4304

<div type='discussions-op-text'>

<sup>Originally posted by **petzolino** February  9, 2025</sup>
Thank you for this fantastic piece of software. Everything is running, including dew heater module etc..  except for the Moon Variables. They are switched on in Layout and App options, come up in Variable Manager as ${MOON_ELEVATION} ${MOON_AZIMUTH ${MOON_ILLUMINATION} but  they are not generated under All Variables as AS_xxxxx.

I've brought them manually with the add variables, are seen under the variable manager as editable variables but still do not show up values. no ??? as values either.

Sun Fields work as expected.

I'd appreciate anyone having a look and maybe resolving this one... thanks !
[support-none-20250209164923.zip](https://github.com/user-attachments/files/18725141/support-none-20250209164923.zip)

</div>

These message appear in `/var/log/allsky.log`:
~~~
2025-02-09T16:20:33.014310-06:00 allskypi5 allsky[74604]: ERROR: Unable to download de421.bsp: cannot download
     https://ssd.jpl.nasa.gov/ftp/eph/planets/bsp/de421.bsp because <urlopen error [SSL: CERTIFICATE_VERIFY_FAILED]
     certificate verify failed: self-signed certificate in certificate chain (_ssl.c:992)>
2025-02-09T16:20:33.014347-06:00 allskypi5 allsky[74604]: INFO: Moon enabled but cannot use due to prior error.
2025-02-09T16:20:33.014417-06:00 allskypi5 allsky[74604]: INFO: Planets enabled but unable to use due to prior error.
~~~

<hr>

## Workaround
If `~/allsky-OLD/config/overlay/config/tmp/overlay/de421.bsp` exists,
copy it to `~/allsky/config/overlay/config/tmp/overlay`

nofootprintz, 2025-02-14:
I don't have the file in my allsky-OLD folder. Is there another place where I could download it from?

Thank you!!

EricClaeys, 2025-02-16:
For anyone that needs it, here's the file.  To use it:
~~~
cd ~/allsky/config/overlay/config/tmp/overlay
# copy the attached file here
mv de421.bsp.txt  de421.bsp
~~~

[de421.bsp.txt](https://github.com/user-attachments/files/18813200/de421.bsp.txt)

ChrisSwallowWL, 2025-02-16:
You can also use "wget --no-check-certificate https://ssd.jpl.nasa.gov/ftp/eph/planets/bsp/de421.bsp" to download the file while ignoring the certificate error.

EricClaeys, 2025-02-16:
@ChrisSwallowWL, thanks, I should have mentioned that for people who prefer to get the file that way.
Eric

oe3nfc, 2025-02-17:
I have the file in the respective directory but still no moon data?

[image]

oe3nfc, 2025-02-17:
[image]

These are the rights of the file

EricClaeys, 2025-02-17:
Hi @oe3nfc,
Are you sure the file is in `~/allsky/config/overlay/config/tmp/overlay` and not in
`~/allsky/config/overlay/config/tmp` ?

If so, please run `grep ERROR /var/log/allsky.log`.

oe3nfc, 2025-02-17:
[image]

Hi Eric, yes!

Saw the file loaded with the wget command shows a date of 2008 - loaded now the file you posted earlier in this thread - but still nada...

Many thanks!

EricClaeys, 2025-02-17:
Your screenshot from 7 hours ago showed it up one level.

Something else must be going in.
Please run `grep ERROR /var/log/allsky.log`

oe3nfc, 2025-02-17:
[image]

Allsky loading without errors -

re your request: 

[image]

aaronwmorris, 2025-02-17:
The correct and long term method for fixing this problem is to ensure the `certifi` python module is up to date.  This module contains the CA certificates needed to validate the server certificates when downloading the ephemeris data.

oe3nfc, 2025-02-17:
Hi Aaron, this beyond my knowledge- can you kindly explain how to achieve this?

many thanks - Norbert

EricClaeys, 2025-02-17:
@oe3nfc, that's something we'll do for the next Allsky release as part of resolving this Issue.

Strange that grep said the log file was a binary file.  Would you please attach `/var/log/allsky.log` so I can check it out?
Thanks - Eric

aaronwmorris, 2025-02-17:
@oe3nfc I am guessing you just activate the allsky virtualenv and upgrade the module there...

```
source venv/bin/activate

pip3 install --upgrade certifi
```

The python requirements.txt files need to be updated with something like the following so the installer will automatically upgrade this module.

```
certifi >= 2024.7.4
```

EricClaeys, 2025-02-18:
@aaronwmorris, thanks a ton!

@Alex-developer, FYI

oe3nfc, 2025-02-18:
> [@oe3nfc](https://github.com/oe3nfc), that's something we'll do for the next Allsky release as part of resolving this Issue.
> 
> Strange that grep said the log file was a binary file. Would you please attach `/var/log/allsky.log` so I can check it out? Thanks - Eric

Ihave tried it but answer is "permission denied"

oe3nfc, 2025-02-18:
Have tried all the suggested tips but still no joy. Nevermind, will wait for next updated version.

Many thanks for all the help!

ChrisSwallowWL, 2025-02-18:
> Ihave tried it but answer is "permission denied"

The log file might be owned by the root account.   Use sudo to view it.

i.e. 'sudo cat /var/log/allsky.log'.

If you wanted to grab the last 100 lines of the log you could do 'sudo tail -100 /var/log/allsky.log'

EricClaeys, 2025-02-18:
@oe3nfc, what command are you using to access the log file?
Not sure how experienced you are with Linux, but a common problem is to simply type
~~~
/var/log/allsky.log
~~~
which results in "Permission Denied", because just entering the name of the file is telling Linux to "execute" the file.  And since the log file is just a file with text in it, and not a program, it fails.

Please run `ls -l /var/log/allsky.log` so I can see what the permissions are.

oe3nfc, 2025-02-18:
[image]

hello Eric, hope this shows what you are looking for?

BobW55, 2025-03-01:
YES. This RESOLVED my overlay issue

EricClaeys, 2025-03-10:
@oe3nfc,  FYI,

[image]

This shows that you DO have permission to view the file, so you likely entered the name of the file (`/var/log/allsky.log`) without telling Linux what you wanted to do with the file (view it, edit it, delete it, copy it, etc.).

If you are still having problems with the moon data not appearing, try running this again:
~~~
grep ERROR /var/log/allsky.log
~~~
The file is recreated every week, so hopefully it'll work this time.

closed 2025-05-19 by EricClaeys