AllskyTeam/allsky #4338: New reporting method

Alex-developer, 2025-02-13:
### Discussed in https://github.com/AllskyTeam/allsky/discussions/4337

<div type='discussions-op-text'>

<sup>Originally posted by **evripidis** February 13, 2025</sup>
As per instructions, we could add the Github discussion number for future reference.

My thread [here](https://github.com/AllskyTeam/allsky-modules/discussions/241) is number 241 in /allsky-modules/discussions/

The thing is that as far as I can see the github button automatically searches the ID only under the /allsky/issues/, so if I click on the "Problem Issue 241" it redirects [here](https://github.com/AllskyTeam/allsky/issues/241).

[image]

</div>

EricClaeys, 2025-03-10:
The "Problem" link assumes it's a Discussion in the "allsky" repository.  Based on the filename (`support-241-20250310010203.zip`) there's no way to determine anything else.

If the filename included the repository and type (Issue or Discussion), e.g., `support-allsky-discussion-241.20250310010203.zip`, the code that creates the link could then use this information to point to the correct place.

EricClaeys, 2025-03-10:
I created branch `4338-new-reporting-method` for this.