AllskyTeam/allsky #4374: Allsky freezes

Empire4191, 2025-02-23:
### DO NOT USE!  For Allsky Team use only

I use the version v2024.12.06
After 55 days of running, the image froze at 6:21 am.

The message in /var/log/allsky.log shows:

2025-02-23T06:17:25.475453+01:00 allsky allsky[2461294]: STARTING EXPOSURE at: 2025-02-23 06:17:25   @ 29.1 sec
2025-02-23T06:17:56.704616+01:00 allsky allsky[2461294]:   > Saving NIGHT image 'image-20250223061725.jpg'
2025-02-23T06:18:32.376038+01:00 allsky allsky[2461294]: STARTING EXPOSURE at: 2025-02-23 06:18:32   @ 24.4 sec
2025-02-23T06:18:59.072442+01:00 allsky allsky[2461294]:   > Saving NIGHT image 'image-20250223061832.jpg'
2025-02-23T06:19:37.824919+01:00 allsky allsky[2461294]: STARTING EXPOSURE at: 2025-02-23 06:19:37   @ 21.4 sec
2025-02-23T06:20:01.407850+01:00 allsky allsky[2461294]:   > Saving NIGHT image 'image-20250223061937.jpg'
2025-02-23T06:20:43.216446+01:00 allsky allsky[2461294]: Processing end of night data
2025-02-23T06:20:43.219576+01:00 allsky allsky[2461294]: ==========
2025-02-23T06:20:43.219888+01:00 allsky allsky[2461294]: === Starting daytime capture ===
2025-02-23T06:20:43.220037+01:00 allsky allsky[2461294]: ==========
2025-02-23T06:20:43.220264+01:00 allsky allsky[2461294]: STARTING EXPOSURE at: 2025-02-23 06:20:43   @ 18.3 sec
2025-02-23T06:20:43.495766+01:00 allsky allsky[3925059]: endOfNight.sh: ===== Posting twilight data
2025-02-23T06:20:48.594982+01:00 allsky allsky[3925059]: endOfNight.sh: ===== Generating Keogram for 20250222
2025-02-23T06:21:03.808541+01:00 allsky allsky[2461294]:   > Saving DAY image 'image-20250223062043.jpg'
2025-02-23T06:21:40.752068+01:00 allsky allsky[2461294]: STARTING EXPOSURE at: 2025-02-23 06:21:40   @ 23.1 sec
2025-02-23T06:21:51.720641+01:00 allsky allsky[3925059]: endOfNight.sh: ===== Keogram complete
2025-02-23T06:21:59.686270+01:00 allsky allsky[3925059]: endOfNight.sh: ===== Generating Startrails for 20250222
2025-02-23T06:22:06.352346+01:00 allsky allsky[2461294]:   > Saving DAY image 'image-20250223062140.jpg'
2025-02-23T06:22:52.104617+01:00 allsky allsky[2461294]: STARTING EXPOSURE at: 2025-02-23 06:22:52   @ 14.3 sec
2025-02-23T06:23:22.389072+01:00 allsky allsky[3925372]: startrails: Minimum: 0.135753   maximum: 0.718018   mean: 0.211106   median: 0.169509
2025-02-23T06:23:22.646084+01:00 allsky allsky[3925059]: endOfNight.sh: ===== Startrails complete
2025-02-23T06:23:26.044290+01:00 allsky allsky[3925059]: endOfNight.sh: ===== Generating Timelapse for 20250222
2025-02-23T06:27:39.867274+01:00 allsky allsky[3925059]: endOfNight.sh: ===== Timelapse complete
2025-02-23T06:27:50.663806+01:00 allsky allsky[3927270]: endOfNight.sh: Deleting old directory /home/admin/allsky/images/20250113
2025-02-23T06:27:51.158654+01:00 allsky allsky[3925059]: endOfNight.sh: ===== Posting camera details to Allsky map.
2025-02-23T06:27:52.646194+01:00 allsky allsky[3927321]:   You can make 4 more updates today.

After a reboot, the AllSky ran normally again.

EricClaeys, 2025-02-24:
I'm curious why you created an Issue when the page said not to?  You even removed the "DO NOT USE" message.

I suspect you have an RPi HQ camera, and you are creating full-size timelapse videos?
If so, decrease the timelapse resolution - I suggest making it half of the sensor size, which is 4056 x 3040.

closed 2025-02-24 by EricClaeys

Empire4191, 2025-02-24:
Thanks for the answer.
Ok sorry, I didn't know that you weren't allowed to report "problems" here?!
Yes, I use the Raspberry HQ camera. The resolution of the time lapse is reduced to half. It worked for many days. Only this one morning the system froze.
Unfortunately, I don't know why.

EricClaeys, 2025-02-24:
@Empire4191,
you said:
> After 55 days of running, the image froze at 6:21 am.

and
> Only this one morning the system froze.

Are those statements referring to the same event?  Is "the system" the Pi, or Allsky, or just the image?

Some people report Allsky freezes at the end of night, but the Pi is still running.  I've never been able to determine if Allsky is really frozen or not, meaning it's running but not doing anything.
I can't recall any cases where the Pi itself froze.  When there's a problem the image will often "freeze", meaning it's never updated.

Can you explain exactly what happened?  Was the Pi still running?  Did an image with a timestamp of 6:21 am on the screen, and it never updated?

A Pi reboot is rarely needed; if it happens again, run `sudo systemctl status allsky` to see if Allsky is still running.  If it is, wait a couple minutes and run that again to see if anything's changed, or Allsky really is "frozen".

In the newest release (v2024.12.06_01) we added a "Getting Support" page to the WebUI that explains how to get support.  We ask that users enter GitHub Discussion items instead of Issues.  We are trying to limit Issues to problems that require a change to Allsky, and most things people report don't require Allsky changes.  Usually they are user error, settings misconfiguration, or other things.

Empire4191, 2025-02-25:
The Raspberry Pi 4 and the AllSky software were accessible. However, it did not show a new photo from the camera.

What was different in the allSky.log was this line:
2025-02-23T06:27:52.646194+01:00 allsky allsky[3927321]: You can make 4 more updates today.

EricClaeys, 2025-02-25:
> You can make 4 more updates today.

You can ignore those messages.  Every other day your map data is uploaded to the Allsky Map server, and it responds with that message.  It only allows 5 updates per day, to minimize the chance of hackers.

If I understand correctly, the only problem you're currently having is the image from the camera doesn't change?
1. Where does it not change?  In the WebUI's "Live View" page, or in the local or remote Allsky Website?
2. Does the image that doesn't change say anything, like "Allsky is starting up"?
3. Please attach these files:  `~/allsky/config/settings.json` and `/var/log/allsky.log`.

Empire4191, 2025-02-27:
1. The image was no longer updated on the website or in the GUI.
2. No, it normally shows the last image that was uploaded.
3. This was the allsky.log which I downloaded a few hours after the outage.

[settings.json](https://github.com/user-attachments/files/19012208/settings.json)

[allsky.log](https://github.com/user-attachments/files/19012181/allsky.log)

EricClaeys, 2025-02-27:
Thanks @Empire4191,

Please refresh your browser when in the WebUI's "Live View" page and when on your remote Website page by pressing SHIFT-F5.  I see your remote Website images being updated, so it's probably a browser caching issue.
If it's still a problem, please change the `Debug Level` to 3 in the WebUI, wait about 20 minutes, then attach `/var/log/allsky.log` again.

I like what you did with the icons on the left side of your remote Website!