AllskyTeam/allsky #4435: Incorrect allskyDefines.inc.repo file

EricClaeys, 2025-03-17:
> When I click on the link in the WebUI to "see the actions that need to be performed" it still shows the following:
> 
> ========== ACTION NEEDED:
> Your remote Website needs to be updated to version v2024.12.06_02.
> It is at version v2024.12.06.     Run:
> 
>   cd ~/allsky;  ./remoteWebsiteInstall.sh
> 
> Note that you do NOT need to manually copy files to the Website;
> the installation does that for you.
> 
> However, my website appears to have been updated to v2024.12.06_02 (the info link shows this version in any case). Also, clicking on the link "to remove this message" does not seem to do anything. I was able to use the "clear messages" button to the same effect however.
> 
>  

 _Originally posted by @sjtaubman in [#4429](https://github.com/AllskyTeam/allsky/issues/4429#issuecomment-2730253821)_

## BUG
Not sure why, but the changes I made to `config_repo/allskyDefines.inc.repo` to add some entries didn't make it into v2024.12.06_02.

EricClaeys, 2025-03-17:
To fix this problem, do the following:
~~~
cd ~/allsky/config_repo
### Copy the attached file here
mv allskyDefines.inc.repo allskyDefines.inc.repo.NG
mv allskyDefines.inc.repo.txt allskyDefines.inc.repo
chmod 644 allskyDefines.inc.repo
cd ..
./install.sh --function update_php_defines
~~~
Next time you get a "click here to ..." message in the WebUI, clicking there should bring up a new browser tab with a short message in it.

[allskyDefines.inc.repo.txt](https://github.com/user-attachments/files/19302259/allskyDefines.inc.repo.txt)

sjtaubman, 2025-03-17:
Thank you again. Made this change.

closed 2025-05-19 by EricClaeys