AllskyTeam/allsky #4442: Website should only display "can't find" message if image.jpg is missing twice in a row.

EricClaeys, 2025-03-21:
### DO NOT USE!  For Allsky Team use only

From [Discussion #4441](https://github.com/AllskyTeam/allsky/discussions/4441).

## Problem
If "image.jpg" isn't found, an Allsky Website will display the "The image at image.jpg is not found or is corrupted" message.  This can legitimately happen in that very small window when the old file is removed and the new one renamed to "image.jpg".

## Fix
Only display the message if at least 2 failures occur in a row.

Flyfisherwiz, 2025-03-21:
Inserted the  .js code as you requested...warning periodically pops up yet.... 

Maybe I need to increase the delay between images?  Currently set at 20000...

[image]

Flyfisherwiz, 2025-03-21:
Changed Debug Level to 3 as requested.  Re-ran allsky-config check_post_data as requested - these are results:

n8yel@allsky:~ $ allsky-config check_post_data

*** Testing postData.sh...
    postData.sh succeeded:
        upload.sh: FTP '/home/n8yel/allsky/tmp/data.json' to dutchghe @ panel.freehosting.com 'public_html/allsky//data.json'
        upload.sh: FTP '/home/n8yel/allsky/config/settings.json' to dutchghe @ panel.freehosting.com 'public_html/allsky//viewSettings/settings.json'

*** postData.sh ran 2 time(s) recently.

*** Checking for recent failures...
    No failures found.

Could not find any problems with postData.sh.
If you think there is a problem, create a new Discussion in GitHub,
describing what messages you've seen.
n8yel@allsky:~ $

The 2 forward slashes seem off to me

Question:  On the WebUi, the 'Websites and Remote Server Settings'  what is the proper syntax format for that??

Flyfisherwiz, 2025-03-22:
Thought maybe you might like/want to review this - I'd be curious to get your thoughts.  Thanks.

[support-none-20250321200013.zip](https://github.com/user-attachments/files/19398773/support-none-20250321200013.zip)

EricClaeys, 2025-03-22:
What am I supposed to look for?

I see:
* You fixed the "type" of several variables.
* You appear to have changed something on March 20 around 3:18 pm because `libpng warning: iCCP: known incorrect sRGB profile` messages started appearing and uploads failed since `/allsky` doesn't exist.
* Around 3:45 you fixed the `/allsky` problem.
* Around 21:19 on the 20th you covered the camera or turned off the lights?  Something happened to significantly darken the images.
* Around 4 am on the 21st whatever happened in the prior item went away.
* Around the same time you started having occasional network problems.

EricClaeys, 2025-03-22:
Also:
* I suggest applying the fix at https://github.com/AllskyTeam/allsky/issues/4435.
* Please run `ls -ld /home/n8yel/allsky/config/modules`.  If it doesn't return this: `drwxrwxr-x 2 pi www-data 4096 ...` do the following: `sudo chgrp www-data  /home/n8yel/allsky/config/modules; chmod 775 /home/n8yel/allsky/config/modules`
* Run `ls -l /home/n8yel/allsky/config/modules`.

EricClaeys, 2025-03-22:
Comments on your earlier post:

You can ignore the double slashes.  Linux treats "//" like "/".  However, the next release will only have one slash.

> Maybe I need to increase the delay between images? Currently set at 20000...

The delay you have on the Pi has nothing to do with the `image.jpg" isn't found` message.  In your browser, please press F12 and in the right side of the screen, go to the "Console" tab.  Next time you get the error message, attach a screenshot of the Console messages.
After that happens, until I correctly fix the problem you might want to increase the `intervalSeconds` in the `remoteConfiguration.json` file in the WebUI's "Editor" page to 10 or 15.  This setting determines how often the browser looks for a new image.  The more often it looks the more likely it'll fail.

Flyfisherwiz, 2025-03-22:
The obscured image was due to me having brought the system indoors and placed on my bench.

The changes you mentioned are memaking a change, undoing a change  'learning on the fly' so to speak.

I ran the ls -ld /home/n8yel/allsky/config/modules, the result came back as drwxrwxr-x 2 n8yel www-data 4096 so that's good.

I'm not asking to you to solve everything 100%, just steer me in the right direction, which you've done (and Thanks) - I don't feel I'm doing too bad for a 60 some odd old geezer who's been out of the IT field (99% MS file & email servers, winders PC's & such in a corproate setting)

EricClaeys, 2025-03-22:
@Flyfisherwiz, I'm also a 60 some odd old geezer but have only been out of IT for about 3 years.  Started my career at Bell Labs working on UNIX (which is what Linux is based on), so I'm very familiar with it.

Sounds like my power of deduction is still pretty good regarding why the images got dark.

> I ran the ls -ld /home/n8yel/allsky/config/modules, the result came back as drwxrwxr-x 2 n8yel www-data 4096 so that's good.

Actually, I was hoping the permissions would be wrong, given a bunch of these messages from the web server:
~~~
PHP Warning:  copy(/home/n8yel/allsky/config/modules/postprocessing_periodic.json-last): Failed to open stream: Permission denied in /home/n8yel/allsky/html/includes/moduleutil.php on line 327
~~~

Please run `ls -l /home/n8yel/allsky/config/modules`.
That might give me an idea why you're getting "Permission denied".

Let me know if the other things I suggest you do helped.

Flyfisherwiz, 2025-03-22:
Good morning.

Ran as requested - results below.

n8yel@allsky:~ $ ls -l /home/n8yel/allsky/config/modules
total 100
-rw-rw-r-- 1 n8yel www-data   370 Mar 18 18:14 module-settings.json
-rw-r--r-- 1 n8yel n8yel      434 Mar 22 08:04 postprocessing_day-debug.json
-rw-rw-r-- 1 n8yel www-data  1769 Mar 16 16:05 postprocessing_day.json
-rw-r--r-- 1 n8yel n8yel        2 Mar 21 20:18 postprocessing_daynight-debug.json
-rw-rw-r-- 1 n8yel www-data     2 Mar 16 16:05 postprocessing_daynight.json
-rw-r--r-- 1 n8yel n8yel        2 Mar 22 07:20 postprocessing_nightday-debug.json
-rw-rw-r-- 1 n8yel www-data     2 Mar 16 16:05 postprocessing_nightday.json
-rw-r--r-- 1 n8yel n8yel      434 Mar 22 07:09 postprocessing_night-debug.json
-rw-rw-r-- 1 n8yel www-data  2169 Mar 18 18:14 postprocessing_night.json
-rw-r--r-- 1 n8yel www-data 18793 Mar 18 17:54 postprocessing_night.json-last
-rw-r--r-- 1 n8yel n8yel      220 Mar 22 08:04 postprocessing_periodic-debug.json
-rw-rw-r-- 1 n8yel www-data 17730 Mar 19 20:23 postprocessing_periodic.json
-rw-r--r-- 1 n8yel www-data 16623 Mar 18 17:21 postprocessing_periodic.json-last

Flyfisherwiz, 2025-03-22:
Created a service, today, to disable the Wifi power management on each re-boot of the Pi to keep the Wifi from going to sleep folowing the instuctions at [this post](https://forums.raspberrypi.com/viewtopic.php?t=337943#p2024004) to improve connectivity - I got to thinking about that this morning, if that could be an issue as I've had instances where I could not connect after a period of time/inactivity.

EricClaeys, 2025-03-22:
Wi-Fi going to sleep definitely would cause issues. I would have thought that upoading a file every couple minutes would keep the WiFi on.

closed 2026-10-13 by EricClaeys