AllskyTeam/allsky #4507: ERROR: Module allsky_overlay.py failed on line 212

leandroguats27, 2025-04-21:
### DO NOT USE!  For Allsky Team use only

no overlay data appears on my allsky
This error message appears in the log
STARTING EXPOSURE at: 2025-04-20 22:19:34   @ 30.0 sec
2025-04-20T22:19:35.747589-03:00 allskycocal allsky[6373]: ERROR: Module allsky_overlay.py failed on line 212 - cannot import name '_imagingft' from 'PIL' (/home/leandro/allsky/venv/lib/python3.11/site-packages/PIL/__init__.py)

Alex-developer, 2025-04-21:
What version of Allsky are you using? If its the latest can you post the support information please

leandroguats27, 2025-04-21:
Version: v2024.12.06_02

2025-04-20T22:18:58.707978-03:00 allskycocal allsky[6210]:      ***** Starting Allsky *****
2025-04-20T22:19:00.783019-03:00 allskycocal allsky[6323]: *****************************************************
2025-04-20T22:19:00.783246-03:00 allskycocal allsky[6323]: *** Allsky Camera Software Version v2024.12.06_02 ***
2025-04-20T22:19:00.783338-03:00 allskycocal allsky[6323]: *****************************************************
2025-04-20T22:19:00.783408-03:00 allskycocal allsky[6323]: Capture images of the sky with a Raspberry Pi and  a ZWO ASI camera
2025-04-20T22:19:00.783481-03:00 allskycocal allsky[6323]: Author: Thomas Jacquin - <jacquin.thomas@gmail.com>
2025-04-20T22:19:00.783553-03:00 allskycocal allsky[6323]: Contributors:
2025-04-20T22:19:00.783623-03:00 allskycocal allsky[6323]:  -Knut Olav Klo
2025-04-20T22:19:00.783701-03:00 allskycocal allsky[6323]:  -Daniel Johnsen
2025-04-20T22:19:00.783769-03:00 allskycocal allsky[6323]:  -Robert Wagner
2025-04-20T22:19:00.783835-03:00 allskycocal allsky[6323]:  -Michael J. Kidd - <linuxkidd@gmail.com>
2025-04-20T22:19:00.783899-03:00 allskycocal allsky[6323]:  -Rob Musquetier
2025-04-20T22:19:00.783967-03:00 allskycocal allsky[6323]:  -Yang and Sam from ZWO
2025-04-20T22:19:00.784027-03:00 allskycocal allsky[6323]:  -Chris Kuethe
2025-04-20T22:19:00.784092-03:00 allskycocal allsky[6323]:  -Eric Claeys
2025-04-20T22:19:00.784179-03:00 allskycocal allsky[6323]:  -Andreas Lindinger
2025-04-20T22:19:00.784698-03:00 allskycocal allsky[6323]:  Camera Information:
2025-04-20T22:19:00.784888-03:00 allskycocal allsky[6323]:   - Type: ZWO
2025-04-20T22:19:00.785079-03:00 allskycocal allsky[6323]:   - Model: ASI662MC (ASI662MC)
2025-04-20T22:19:00.785336-03:00 allskycocal allsky[6323]:   - ID: [none]
2025-04-20T22:19:00.785718-03:00 allskycocal allsky[6323]:   - Serial Number: 041a0e0a17020900
2025-04-20T22:19:00.785796-03:00 allskycocal allsky[6323]:   - Native Resolution: 1920x1080
2025-04-20T22:19:00.785860-03:00 allskycocal allsky[6323]:   - Pixel Size: 2.90 microns
2025-04-20T22:19:00.785924-03:00 allskycocal allsky[6323]:   - Supported Bins: 1 2
2025-04-20T22:19:00.785986-03:00 allskycocal allsky[6323]:   - Color camera, bayer pattern: RG
2025-04-20T22:19:00.786443-03:00 allskycocal allsky[6323]:   - Sensor temperature: 37.6 C
2025-04-20T22:19:00.786641-03:00 allskycocal allsky[6323]:   - Bit depth: 12
2025-04-20T22:19:00.973174-03:00 allskycocal allsky[6323]: Settings:
2025-04-20T22:19:00.973307-03:00 allskycocal allsky[6323]:    Image Type: RGB24 (1)
2025-04-20T22:19:00.973374-03:00 allskycocal allsky[6323]:    Resolution (before any binning): 1920x1080
2025-04-20T22:19:00.973486-03:00 allskycocal allsky[6323]:    Configuration file: /home/leandro/allsky/tmp/capture_args.txt
2025-04-20T22:19:00.973550-03:00 allskycocal allsky[6323]:    Quality: 95
2025-04-20T22:19:00.973612-03:00 allskycocal allsky[6323]:    Daytime capture: No
2025-04-20T22:19:00.973673-03:00 allskycocal allsky[6323]:    Daytime save: No
2025-04-20T22:19:00.973736-03:00 allskycocal allsky[6323]:    Nighttime capture: Yes
2025-04-20T22:19:00.973796-03:00 allskycocal allsky[6323]:    Nighttime save: Yes
2025-04-20T22:19:00.973864-03:00 allskycocal allsky[6323]:    Exposure (day):   500.00 ms (0.50 sec), Auto: Yes, Max Auto-Exposure: 60.0 sec
2025-04-20T22:19:00.973946-03:00 allskycocal allsky[6323]:    Exposure (night):        30.0 sec, Auto:  No
2025-04-20T22:19:00.974008-03:00 allskycocal allsky[6323]:    Gain (day):       0, Auto: Yes, Max Auto-Gain: 300
2025-04-20T22:19:00.974068-03:00 allskycocal allsky[6323]:    Gain (night):   250, Auto:  No
2025-04-20T22:19:00.974130-03:00 allskycocal allsky[6323]:    Gain Transition Time: 5.0 minutes
2025-04-20T22:19:00.974198-03:00 allskycocal allsky[6323]:    Target Mean Value (day):       0.500
2025-04-20T22:19:00.974259-03:00 allskycocal allsky[6323]:    Target Mean Value (night):     0.300
2025-04-20T22:19:00.974321-03:00 allskycocal allsky[6323]:    Target Mean Threshold (day):   0.020
2025-04-20T22:19:00.974381-03:00 allskycocal allsky[6323]:    Target Mean Threshold (night): 0.020
2025-04-20T22:19:00.974441-03:00 allskycocal allsky[6323]:    Binning (day):   1
2025-04-20T22:19:00.974501-03:00 allskycocal allsky[6323]:    Binning (night): 1
2025-04-20T22:19:00.974561-03:00 allskycocal allsky[6323]:    White Balance (day)   Red: 55, Blue: 75, Auto:  No
2025-04-20T22:19:00.974627-03:00 allskycocal allsky[6323]:    White Balance (night) Red: 55, Blue: 75, Auto:  No
2025-04-20T22:19:00.974705-03:00 allskycocal allsky[6323]:    Delay (day):   30.0 sec
2025-04-20T22:19:00.974773-03:00 allskycocal allsky[6323]:    Delay (night): 1.0 sec
2025-04-20T22:19:00.975175-03:00 allskycocal allsky[6323]:    Consistent delays: Yes
2025-04-20T22:19:00.975246-03:00 allskycocal allsky[6323]:    Skip Frames (day):   5
2025-04-20T22:19:00.975308-03:00 allskycocal allsky[6323]:    Skip Frames (night): 0
2025-04-20T22:19:00.975370-03:00 allskycocal allsky[6323]:    Aggression: 85%
2025-04-20T22:19:00.975432-03:00 allskycocal allsky[6323]:    USB Speed: 50, auto: Yes
2025-04-20T22:19:00.975506-03:00 allskycocal allsky[6323]:    Flip Image: none (0)
2025-04-20T22:19:00.975600-03:00 allskycocal allsky[6323]:    Filename: image.jpg saved to /home/leandro/allsky/tmp
2025-04-20T22:19:00.975670-03:00 allskycocal allsky[6323]:    Latitude: 3.4686S, Longitude: 41.5578W
[... cut ...]
2025-04-20T22:25:09.090902-03:00 allskycocal allsky[7148]:   > Image took 122.0 ms to save (average 134.0 ms).
2025-04-20T22:25:09.096556-03:00 allskycocal allsky[7316]: saveImage.sh NIGHT /home/leandro/allsky/tmp/image-20250420222436.jpg DATE=20250420 TIME=222436 AUTOEXPOSURE=0 sAUTOEXPOSURE= EXPOSURE_US=30000000 sEXPOSURE=30.0 sec AUTOGAIN=0 sAUTOGAIN= GAIN=250 AUTOWB=0 sAUTOAWB= WBR=55 WBB=75 MEAN=0.210938 TEMPERATURE_C=38 TEMPERATURE_F=100 BIN=1 FLIP=none BIT_DEPTH=8 DARKFRAME=0 eOVERLAY=1 AUTOUSB=1 USB=50

Alex-developer, 2025-04-21:
Thanks but can you please use the 'Getting support' option in the main menu and post the results of that, this will help determine whats installed

leandroguats27, 2025-04-21:
[support-none-20250420222506.zip](https://github.com/user-attachments/files/19828330/support-none-20250420222506.zip)

Alex-developer, 2025-04-21:
Try the following

```
sudo apt update
sudo apt install libfreetype6-dev libjpeg-dev zlib1g-dev

source /home/leandro/allsky/venv/bin/activate
pip uninstall Pillow
pip install --no-cache-dir --force-reinstall Pillow
```

That will force a reinstall of pillow

EricClaeys, 2025-04-21:
@leandroguats27,
When you created this Issue did you not see the **DO NOT USE! For Allsky Team use only** message?

I'm converting this to a Discussion.