AllskyTeam/allsky #4512: Delete support for "legacy" overlay method

EricClaeys, 2025-04-23:
### DO NOT USE!  For Allsky Team use only

v2024.12.06_02 advertised itself as the last Allsky release to support the legacy overlay method.

This requires deleting the associated settings in the WebUI, removing that code from capture_{RPi|ZWO}, removing the check for the legacy method in the Overlay Editor.

lkomage, 2025-05-16:
How will this affect the display of the histogram box? Looks like it still requires things to be set to legacy to be able to see it on the live camera view.

Alex-developer, 2025-05-16:
We will make sure it doesnt

EricClaeys, 2025-05-17:
@Alex-developer, the code to display the histogram box is with the other "legacy" code in the "capture" programs that will be deleted.
In v2025.xx.xx_01 I plan to replace the histogram box with an optional user-generated mask that defines the auto-exposure area (like the RPi has) so I suggest for v2025.xx.xx we do something very easy and temporary to add the histogram box to the image.  Perhaps a module that reads the histogram size and location settings?  It will also need the sensor and image size.