AllskyTeam/allsky #4525: Adjustable start / stop times for timelapse, keogram, and startrails

EricClaeys, 2025-05-02:
### Discussed in https://github.com/AllskyTeam/allsky/discussions/4516

<div type='discussions-op-text'>

<sup>Originally posted by **gert4git** April 26, 2025</sup>
Hello All,

Been discussing this in the Q&A, I want to send this as a feature request just to have it 'on the books'. :-)

Suggestion is for an enhancement to set the start time for the 24h timelapse video produced in Allsky. To be added as an item under "Timelapse Settings". Have a user entry for start time of 24h timelapse. Enter any 24hr time to be starting point. Maybe some keyword entries like 'local midnight' / 'local noon' / dusk / dawn / sunrise / set. 

Thank you for your consideration.

Cheers,
Gert
</div>

## Details
Allow the user to set a start date/time and/or end date/time for timelapse, keogram, and startrails.
Each item would have an optional start time and end time.  If both are blank, "yesterday" will be used as is currently done.
The `generateForDay.sh` command will look for the start/end time settings and create a file that contains every image file name between the two times, then pass that file to `timelapse.sh` or use it when creating the keogram or startrails.

## To figure out
The `timelapse.sh` command already accepts a list of files to process.  Creating keograms and startrails will need to as well.
* These settings should probably only apply when the timelapse/keogram/startrails are created via a new module, not when the items are created at the end of nighttime like is currently done.
* Should the settings be in the WebUI's "Allsky Settings" page, or in the associated new modules?  Eric's preference is on the "Allsky Settings" page with the other settings, otherwise there's no central place to look at all the non-overlay settings. 
* Should only the TIME be set by the user or the TIME and DATE?
If only the TIMEs are specified, should we assume they are yesterday, e.g., 12:00:01 am to 11:59:59 pm to get a timelapse for yesterday?  This implies the module needs to run after the end time.  More generally, should the module assume the end time (if specified) has already passed?
* If the user can specify the DATE, should we allow an absolute date like "May 1, 2025" (which probably isn't very useful as a setting though), or "`X` days ago", for example `1 day ago`.  Eric is leaning toward NOT having the DATE in the WebUI settings but allowing it on the `generateForDay.sh` command.  I think adding the date to the WebUI makes it too complex and will likely rarely be used.  If someone wants a timelapse containing multiple days they can manually call `generateForDay.sh`.
* We should probably allow absolute times, e.g., 12:00:01 as well as some aliases, like "daytime_start", "daytime_end", "nighttime_start, "nighttime_end", "sunrise", and "sunset".  The current algorithm would then be:  Start Time: "daytime_start", End Time: "nighttime_end".
* Should the user be able to set only the start OR end time and leave the other blank?  If so, need to determine what that means, e.g.,:
  * If only the start time is given, when should the end time be?
  * If only the end time is given, when should the start time be?

benhartwich, 2026-10-02:
@EricClaeys I'd like to take this on for v2026.10.01_01. Here's a proposal for the open questions, so we can agree on the design before I write it.

**Settings:** on the Allsky Settings page, as you prefer. Each of Timelapse, Keogram and Startrails gets a *Start time* and an *End time*, time only, no date. Allowed values:
- blank: today's behaviour;
- `HH:MM`;
- an alias: `sunrise`, `sunset`, `daytime_start`, `nighttime_start`. The last two are the current day/night transitions, so the default is the same as blank.

If only one is set, the window runs from it, or up to it, over the 24 hours before the end. So start `12:00` alone means noon to noon.

**Which images:** the most recent window that has fully ended. Images are picked by the time in their file name (`image-YYYYMMDDHHMMSS`), so a window can cross midnight and span two day folders. The list goes to the existing `--images` option. That's what `timelapse.sh` and `startrails` already accept and what your night-only startrails uses.

**When it runs:**
- **No times set:** unchanged, at the end of night.
- **Times set:** at the first day/night transition after the window's end time (`endOfNight.sh` or `endOfDay.sh`), so no new timer is needed. For noon to noon that's the next sunset.
- The output is named after the day the window ends.

**generateForDay.sh:** new `--start` and `--end` options that take a date and time, e.g. `--start "2026-10-01 12:00" --end "2026-10-02 12:00"`. Dates only on the command line, as you suggested.

**Keogram:** `keogram.cpp` gets `--images`, like `startrails.cpp` has, which removes the "does not (yet) work with keograms" restriction.

Does that match what you had in mind? In particular: should the times apply to the end-of-night generation too, as above, or only to a separate module?

EricClaeys, 2026-10-02:
I think only images in a single allsky/image/YYYYMMDD directory should be use, since that's consistent with everything else.  This means, for example, if a user specifies a start and/or end date, their output file(s) will represent less than 24 hours.

A startrails start time of "nighttime_start" would replace the "nighttime only" setting.  That'll be the shortest-lived setting ever!!

I'd vote to have it always run via `endOfNight.sh` (or manually via `generateForDay.sh`).

`generateForDay.sh` already requires the name of an images/YYYYMMDD folder, so the date isn't needed with the `--start` or `--end` arguments.  **Or were you thinking they could be used to specify output that includes multiple days?**

Needs to work with keolapse if there's a way for a user to create one manually, e.g., via `generateForDay.sh`.

I noticed the word `module` in the original text and don't remember what I was thinking.  Alex and I have talked about putting all the `endOfNight.sh` functionality into multiple modules (which I think is why we have a "keograms" module), but we haven't done anything about it.

benhartwich, 2026-10-03:
@EricClaeys agreed on all points. Updated plan:
- **One folder only:** the times select images within one `images/YYYYMMDD` folder, like everything else. A window that starts later or ends earlier simply gives output for less than 24 hours. No multi-day output.
- **Startrails:** a start time of `nighttime_start` replaces the "nighttime only" setting. I'll migrate an existing "nighttime only = yes" to it, so nobody loses their setting.
- **When it runs:** always from `endOfNight.sh`, or manually with `generateForDay.sh`.
- **`generateForDay.sh`:** `--start` and `--end` take times only (or the aliases), since the folder already gives the date.
- **Keolapse:** the list of selected images is the same `--images` file `generateForDay.sh` already writes. If Keolapse can read such a list, it gets the same window. I'll check with the module's author.
- Moving the end-of-night work into modules can come later; this doesn't depend on it.

EricClaeys, 2026-10-03:
Sounds great.

benhartwich, 2026-10-03:
@EricClaeys it's up in #5303, against `v2026.10.01_01`, as planned above: one folder only, `--start`/`--end` in `generateForDay.sh`, and "Night images only" becomes `nighttime_start`/`nighttime_end` (converted on upgrade). The keogram now also takes an image list (`--images`), so all three use the same selection.

NiteRide, 2026-10-21:
this is pretty cool.  maybe for v_02 should allow/include parameters for standard twilights.
For example running the time-lapse from nighttime_start to nighttime_end, but startrails from astronomical_twilight to astronomical_dawn to avoid brighter sky without relying solely on brightness settings.

NiteRide, 2026-10-21:
Note: at present, if a user is creating a keolapse, the video and keolapse ring will not be in sync if they use different config for start and end times rather than all images. (current module implementation the video will just use all images and the generated keogram) 

I'm not sure how likely this might be, but now that the settings are available there is a reasonable chance they could be set differently based on a users preferences.  

The logical fix is to also update the keotimelapse module so that:
1. Update module to use/pass timelapse start and end parameters rather than using whole folder of images as per current (probably modifying get_source_images function?)
2. check if [keogram start/stop] <> [timelapse start/stop] then trigger creation of a special (temporary?) keogram using the timelapse start/stop to produce the keolapse video.
3. A (probably appropriate) bonus is adding module only settings for Start Time and End Time so that the keolapse could be a different timespan (or timespan entered in the module).  e.g if the user is creating two timelapses or using the other module settings.

I definitely wont have time to built that before the _01 drop.  I created a new issue in the modules repo though to track this adjacent update to adjustable start/stop times: 
https://github.com/AllskyTeam/allsky-modules/issues/418

closed 2026-10-28 by Alex-developer