AllskyTeam/allsky #4529: No response from Webui when clicking that you want to remove the _OLD folder after upgrade

adekloet, 2025-05-04:
### DO NOT USE!  For Allsky Team use only

When I click to to remove the _OLD folder after upgrade I get an empty website response back with no information if an action was performed.

Message on the page after upgrade:

[image]

Click on the message "Click here to remove" gives the following output on the google chrome browser:

[image]

Can a response be added if the action was performed successfully? (Maybe also stay on the same page (not load a new one and reload the webui after the action with a new message that it was removed or that there was a failure?)

EricClaeys, 2025-05-04:
@adekloet,
In the future, please don't create Issues per the **DO NOT USE! For Allsky Team use only**.

What you reported is a known bug that's been fixed in the next release.  It displays a simple message.

Regarding staying on the same page, I asked @Alex-developer to modify the WebUI so I can put the message on the same page.  Having the message on a different page was easy to do, but not very user friendly.

closed 2025-05-04 by EricClaeys

adekloet, 2025-05-04:
> [@adekloet](https://github.com/adekloet), In the future, please don't create Issues per the **DO NOT USE! For Allsky Team use only**.
>
> What you reported is a known bug that's been fixed in the next [...]

Apologies, I missed the "Allsky Team only" won't happen again sir.