AllskyTeam/allsky #4670: Error installing Adafruit-Blinka on Buster in v2024.12.06_04

EricClaeys, 2025-06-21:
### DO NOT USE!  For Allsky Team use only

@Alex-developer 
I just installed the pre-release of v2024.12.06_04 on a Buster Pi 4 and go this:
~~~
    === Package # 11 of 14: [Adafruit-Blinka]

*** ERROR: 
**********
Python_dependencies [Adafruit-Blinka] failed
**********

* The full log file is in /home/pi/allsky/config/logs/Python_dependencies.11.log
The end of the file is:
	      swig -python -o lgpio_wrap.c lgpio.i
	      unable to execute 'swig': No such file or directory
	      error: command 'swig' failed with exit status 1
	      [end of output]
	  
	  note: This error originates from a subprocess, and is likely not a problem with pip.
	  ERROR: Failed building wheel for lgpio
	  Running setup.py clean for lgpio
	Failed to build lgpio
	ERROR: Could not build wheels for lgpio, which is required to install pyproject.toml-based projects
~~~

Alex-developer, 2025-06-21:
Looks like 'swig' is missing. I dont have a pi that I can install Buster on anymore, the more recent Pi's will not allow it to be installed due to firmware incompatibility issues.

I suspect you are right in that the adafruit packages probably have been changed recently

Can you try

`sudo apt install swig`

the re do the installer and see if it sorts it

Alex-developer, 2025-06-21:
Try

` sudo apt-get install lgpio`

Then reinstall