AllskyTeam/allsky #4749: Implement keolapse

EricClaeys, 2025-07-17:
### DO NOT USE!  For Allsky Team use only

Implement a "keolapse" which is a timelapse with a circular keogram around it.

NiteRide, 2025-07-20:
Okay- so one of my ideas is to make it easier to get the keo-ring in the right place on the image.  The setup is quite fiddly and tedious using periodic flow and checking and changing image and settings.  Do-able, but a barrier to entry. 

I envision something like this “keolapse-sizer” that people could use to get close in one step before testing out other parameters:  https://allskywelly.great-site.net/allsky2/keolapse_sizer.php. (I don’t think this works on a mobile browser). Basically:
1. load an image
2. drag around the circle and resize with corner until it’s in pretty much the right spot
3. type the numbers into the module setup
4. test and see, etc

 It might make more sense to be hosted somewhere rather than a part of the module itself?
Also, that link is very bare bones, it might be nice if the fields were input-able not just using the single drag handle. 
*note, although I do have some ability to review or create code, I am not php nor web developer.  I was able to do this with some prompt engineering, then tweaking bits.  It “works” as is, but could use some refinement, maybe I’ll keep at it a little…
Thoughts?

EricClaeys, 2025-07-21:
Do you have a Discord login, or can you create one?  I'd like you to join the Allsky Discord channel so we can discuss your ideas with Jim, the developer of the allsky_keolapse module.

1. `allsky/tmp` IS in memory `keolapse*_temp.mp4 ` should be built there.  I'm surprised I didn't catch it, but am glad you did.
For **new** installs the size is 75 MB for Pi's with 0.5 GB memory and it goes up to 400 MB for Pi's with 8 GB memory.
For **upgrades**, the current size is kept, but I have a comment in the code to `# TODO: set to new, larger size if not already there.`
Increasing 75 MB to 100 MB would probably be ok.

2. Hmmm.  The answer has some far-reaching consequences - glad you asked.  I originally was thinking the WebUI would have a new setting to "Build a keolapse", so a user could build it AND build a normal timelapse.  But that would require changes to the Allsky Website to add another icon on the left to view keolapse, and all the associated changes that implies, as you mentioned.  I'd vote to **initially** to make it an either / or thing and just a new setting to make the timelapse a keolapse instead.  The file name would not change.

> I think modules run after end_of_night.sh?

Correct.  The capture_??? programs (the ones that actually take the images) calls `endOfNight.sh` when transitioning from nighttime to daytime.  It creates the timelapse, keogram, etc. and removes "old" files, THEN invokes the `Night to Day` module flow.  It would need to check if a keolapse is being created, then not create the normal timelapse.

Some people have asked for the option to not keep ANY images at the end of night.  Currently a `Days To Keep` of `0` mean "keep ALL images forever" and that would probably change to `-1` means "keep ALL images forever" and `0` means "don't keep any images".
Assuming we implement that, `endOfNight.sh` would also need to make sure it doesn't remove the prior day's images if a keolapse is being created.  Maybe the fix is to just delete the "old" images AFTER running the modules.

NiteRide, 2025-07-21:
> Do you have a Discord login, or can you create one? I'd like you to join the Allsky Discord channel so we can discuss your ideas with Jim, the developer of the allsky_keolapse module.

I haven’t used Discord in the past, but you can now reach me at jkc523.  Happy to help out, and noting that like most everyone else, this is a hobby so commitment will come and go as life allows!

EricClaeys, 2025-07-23:
@AndyOfLinux,

Thanks for your comments.  Hope you don't mind that I made them into a list - it was hard to read.

Could you describe more your "dynamic keolapse" idea?

> The docs say that "periodic" module should be used

You only want to enable the keolapse in "Periodic" flow while testing.  By default a keolapse will then be created every 60 seconds (in addition to the one at the end of night) until you disable the Periodic flow.
I suspect Jim added this because it's an easy way to force a keolapse to be created without having to manually enter some command.

@Alex-developer, in v2025.xx.xx, modules will have a "Test" button.  Can that be used to create a keolapse?  And then can the "Test" button be pressed multiple times?
If so, then the "Periodic" stuff in the keolapse module can be removed, which will make @AndyOfLinux happy!!

Alex-developer, 2025-07-23:
> [@Alex-developer](https://github.com/Alex-developer), in v2025.xx.xx, modules will have a "Test" button. Can that be used to create a keolapse? And then can the "Test" button be pressed multiple [...]

No ...... The test function doesn't support any form of image manipulation.

What this needs is the similar logic to the 'Mask Editor' thats coming in the next release, the one I briefly showed you last night.

I would add a new field type for modules that allows the keogram circle to be drawn and then generated to test it, so it can be done in real time. Once happy the relevant values would be added to the modules settings. 

This is an older video of how the mask editor works but I think you should be able to get the idea

https://github.com/user-attachments/assets/384edbc2-6da8-43ff-a92b-c227a61654f7