AllskyTeam/allsky #4786: Startrails - improve use and debugging

EricClaeys, 2025-07-27:
### DO NOT USE!  For Allsky Team use only

Startrails work for most people, but some people struggle getting trailed stars.  The [startrails documentation](https://htmlpreview.github.io/?https://raw.githubusercontent.com/AllskyTeam/allsky/master/html/documentation/explanations/startrails.html) along with the `allsky-config get_brightness_info` command help users determine what `Brightness Threshold` to use, but some people still struggle.

This Issue is a catch-all for ways to improve startrails generation and make it even easier to find the right settings to use.

- [ ] Add an option to pass a list of images to process, rather than process all images for a 24-hour period.  A correctly-set `Brightness Threshold` should eliminate daytime image, but only looking at night images will make processing go faster (possibly twice as fast).  Accepting a list of images would also make it easy to implement things like a user-specified start and stop time for image to look at, or using images from multiple nights or directories.  The images in the list would default to a specified directory unless the image name began with `/` in which case it would signify a full pathname to the image.
- [ ] Add an option in the WebUI to only look at nighttime images.  This would use the capability above to create a list of nighttime-only images.  The default should probably be "Yes", although in general I don't like adding things that "break" existing functionality, which this could do if a user's startrails used some daytime images due to their Angle, Latitude, and Longitude.  I think this is not very likely though.
- [ ] Add a "--run-test" argument to `allsky-config get_brightness_info` that would cause it to create a list of 50 or so nightime images, then call the `startrails` command with a bunch of different Brightness Thresholds using that image list, so a bunch of different startrails were created for the user to look at.  It could even let the user know which setting resulted in the most images being used.   The `--run-test` argument would basically automate what the startrails documentation tells the user to do manually.  Maybe use an initial Threshold of the average "Minimum" value of prior startrails, and increment by 0.01 up to twice the "Maximum" value?  Would likely need to play around with these numbers, and possibly let the user specify the number of test startrails created.
- [ ] When created at end of night, if no images were used, i.e., none were below the threshold, the `startrails` command should exit with code EXIT_PARTIAL_OK which results in a System message suggesting the user run `allsky-config get_brightness_info --run-test`.

ova103, 2025-07-28:
Hello
I like the first one (why would you use day time images for startrail). Your concern you have with this one is solved if you choose to keep the day images for startails. I thinks limiting to night images only will solve most of the problems. 

I also like the last one, but I would add a file with the list of all images of the day/night, indicate if they have been used or not, and add the average brightness (and treshold brightness)

Another idea is to add on the overlay a field that mentions if the image is used for the overlay and a filed for the average brightness of the image (and the treshold brightness) . So you can scroll through your images and learn from it. 

Hope this helps.

EricClaeys, 2025-07-28:
@ova103, thanks for the comments.
> why would you use day time images for startrails

In Allsky, "daytime" and "nighttime" are defined by the `Longitude` and `Angle` and define when the "Daytime" and "Nighttime" settings should be use, and are only loosly the same as what we generally think of as "daytime" and "nighttime".
I'm helping a user with an unrelated problem and noticed his `Angle` is -12 which means it's what most people would call "dark" at the beginning and end of his "daytime" for around half an hour, and images in those periods probably contain valid stars for a startrails.  Using only "nighttime" image means he'd probably get an hour's worth less stars in his startrails.
It's cases like this that make me hesitant to have the default be "Use only nighttime images".

To implement the first change I'll probably have the `saveImage.sh` script, which does all the work after a raw image is saved like cropping, adding the overlay, and uploading the file, save the name of the image in a "DAY" or "NIGHT" file.  If only nighttime files are included in a startrails, then the "startrails" command would be given the "NIGHT" list to process.

<hr>

For the last one, I already changed the "startrails" command in the next release to include the Brightness Threshold in it's output along with the Minimum, maximum, etc.  When you said:
> indicate if they have been used or not, 

do you mean "indicate if nighttime only, or all images" are used, or indicate for each image the "startrails" command looks at, whether or not it is included in the startrails, i.e., if it's above or below the threshold?  I considered the later, but if a user has 3000 images that's a lot of data to keep.  For debugging I can see where it would be useful to have an option to the "startrails" command that says "for each file, indicate if it was used or not".

NiteRide, 2025-07-29:
perhaps it may be neat to be able to exclude images that have airplane light trails?  

Don't need to solution it here, but maybe could be implemented by detecting lines across an image?  not sure it a negative or not, but could potentially end up also excluding satellite or meteors as well, but since you are collecting ideas...

jcauthen78, 2025-08-01:
As one of the people who have had troubles with Startrails after the new exposure algorithm, I've mused about it for a while and tried different methods of sorting what may be cloudy and clear images. I'm a non-typical user, as some know. I capture the daytime images into `images/YYYYMMDD/d` so I can review them for interesting weather - have caught a few interesting events using this method. I don't store the daytime images in the core date folder because it'd be over 4,500 images most day + nights. and it makes it easy to delete the daytime only images if they're not of use. 

I've thought about adding _day_ or _night_ to the image names, but 4,500+ images in a single folder is quite troublesome. So brute-forcing the /d folder in the saveImage.sh for the daytime images works for my needs/wants.

I thought about methods of star detection, sky clarity, or other methods and wondered if the Starcount module could help. Since it has been working successfully on my system for quite a while, I figured it may be a metric that may work for 'is this image something that is good for startrails'. 

I spun up a rough metadata module, writing to a JSON file in the previously mentioned ${ALLSKY_IMAGES}/DATA folder. 

Will let it run over night and see what I can glean from the data-grab I have it doing - hopefully the weather is semi-clear to at least grab a few frames with fair starcounts... Angle, Starcount, Mean Brightnes, and whatever else might be useful is grabbed, I know it's TOO much data to collect, but for testing, figured "if its there, grab it". 

[image]

Working on a separate python script (probably module as well) that will dig through the JSON file, do a quick `if "day_or_night" = "NIGHT", and  "starcount" > 90, add file name to list ${ALLSKY_TMP}/starcount_trails.txt`, then try and feed it into generateForDay as a `--images` and see what happens. Comparing it to the current startrails. I'll probably modify the generateForDay.sh a bit to get the --files working with the txt file, but should be able to farmer it together.

Not sure if this is going to work the way my mind wants it to, or even work at all...  BUT - what are y'all's thoughts on a method like this? Anything I should try? or consider in the testing?

EricClaeys, 2025-08-01:
@jcauthen78, Jim,
Thanks a ton for doing this!

I think this will most likely work and could easily form the basis for a script that checks multiple criteria - mean brightness, star count, cloud coverage, day/night, etc.

Some things to take into account when creating it:
1. Assume it will use the database when we know how to use it, so try to compartmentalize references to the ${ALLSKY_IMAGES}/DATA folder so we can replace just that code.
2. Assume there will be multiple criteria that could be used, and each will have a setting or something similar that determines if that criteria is actually used.  For example, if a user doesn't have the Cloud Detection installed, the code would not include that criteria.
3. If there are NO criteria that can be applied, then the list of files that `startrails` looks at would be the current list, which is everything.

You will probably want to install the newest `startrails` command since it accepts a "--images file_name" argument which you'll want to use.  All the criteria checks should result in a file that contains image names for `startrails` to look at.  To get the file:
~~~
cd ~/allsky/src
mv startrails.cpp startrails.cpp- ORIGINAL
# in GitHub, change from the `master` branch to `v2025.xx.xx` and go to the `src` directory
# click on the `startrails.cpp` file
# Use one of the icons to download the file to the directory you are in.
make
cp startrails ../bin
~~~

craigim, 2025-08-02:
Echoing some bullets from the other discussion thread (https://github.com/AllskyTeam/allsky-modules/discussions/276):
 
* Agree that having the option to specify darkness level (civil, nautical, astro) makes a lot of sense to me. It could be a simple dropdown with an "other" option for power users to specify a different angle. Only near the equator does a fixed time really make sense. I think that people who are looking for specific times every night are probably power users rolling their own scripts.
* Because the auto-level algorithm shoots for a fixed brightness by adjusting exposure time and gain, and it take a few frames for responses to changes in brightness, the fixed image brightness is not a good measurement of how dark it is outside. You may well have night images that are brighter than day images, or cloudy images at night that are similar to clear. If you calculate brightness after dividing by the gain and the exposure time, that would normalize out those changes and give you a much more physical value of the scene brightness.
* Being able to provide a mask for where to determine brightness is pretty key in my mind.

EricClaeys, 2025-08-02:
@craigim, thanks for chiming in.
> If you calculate brightness after dividing by the gain and the exposure time, that would normalize out those changes and give you a much more physical value of the scene brightness.

Could you give an example with numbers on what you mean?  And how that's better than just comparing brightness?
Use a nighttime exposure of 60000 ms (60 sec) and gain of 10 and daytime exposure of 0.5 ms and gain of 1.

Assuming a user can specify the start and stop times, e.g., via angles, should a brightness check ALSO be done?

> Being able to provide a mask for where to determine brightness is pretty key in my mind.

For use by the auto-exposure algorithm during image capture, or by the `startrails` command?
Would a user ever want to specify different masks for brightness for the auto-exposure algorithm and the `startrails` command, and if so, often enough to add two new options in the WebUI?
Offhand I would think one mask during image capture is enough, and that mean brightness could be passed to the `startrails` command.

craigim, 2025-08-02:
Here are part of three images. One from noon and two from around midnight on a clear and a cloudy night. The brightness values *are* different, and it is certainly possible to thread the needle to get just the dimmest sky, but to get that I had to push my daytime target threshold way up and my night threshold down.

| Image | Brightness | Gain | Exposure | Normalized Brightness |
| --- | --- | --- | --- | --- |
| Day 1 | 0.482 | 1.00 | 175e-6 seconds | 2777 |
| Night 1 | 0.224 | 1.51 | 20 seconds | 9.26e-3 |
| Night 2 | 0.371 | 1.00 | 10.7 seconds | 3.47e-2 |

Using the normalized brightness, daytime is 7 orders of magnitude brighter than night as expected. No ambiguity, no careful trying to set the target brightness to get it just right. I can also now push the daytime down and the gain up. Certainly you could also use the time, but there is no danger here of a daytime image slipping in.

There is also a lot more room between the cloudy and clear night because on the cloudy night, it only took 11 seconds to get to my set point, but on the dark night, it hit my limit of 20 seconds and bumped up the gain to get over the minimum threshold. If I had set my tolerances tighter, the auto-exposure would have pushed the gain up until the brightness was closer to 0.35. At that point, using brightness alone, I would not have been able to separate them and filter out the clouds. The normalized value, however, would not change by much.

Using a mask to calculate those brightness numbers to just the sky will further separate the cloud/clear skies because the pixels outside the mask don't really depend on exposure time. Also, it's hard to provide advice to users on what values to set because every camera and lens is a little different, so the amount of black border and horizon is different. If limited to just sky values, then the numbers across cameras should be more consistent and independent of the gain and target brightness settings in each setup. 

[image]

[image]

[image]

EricClaeys, 2025-08-02:
Thanks @craigim, this is good and helpful discussion.

If you were to prioritize all the improvements and pick just 1 or 2 for the next release and plan other improvements for follow-on releases, what would you pick?
Keep in mind that we'll almost certainly have a database in the next release where we can easily store per-image data like image name, night/day flag, exposure, mean brightness, gain, and possibly optional data like star count and cloud coverage.  The startrails command will be able to accept a file name that contains a list of images to process (and they can be in multiple directories).  All the intelligence about which images to include in the startrails could be contained in the script that creates the list of files (probably in generateForDay.sh or called by generateForDay.sh).
@Alex-developer can correct me if I am wrong, but I am pretty sure the WebUI will have a way for users to create a mask, like to mask out everything but the sky.  The RPi auto-exposure algorithm currently creates a mask internally that's (I think) a circle centered on the sensor and is 2/3 the height of the sensor.  It would be a fairly easy change to allow it to read in a custom mask instead.
The ZWO algorithm looks at the part of an image defined by the "Histogram box", but it doesn't use a mask so would take a non-trivial effort to read in a mask instead.

Are you willing to help with these improvements?

jcauthen78, 2025-08-03:
@EricClaeys, thank you for the updated scripts. 

After two nights of logging the image capture data, and sorting it with a periodic test loging module, is successfully finding images that are more clear and dark than the default threshold using the Starcount metric. I'm attaching the stock version of startrails, as well as my sorting of 75 stars & 80 stars.  The script still calls the threshold value, haven't figured out if there's a way to tell it to bypass the threshold, but its all good. 

75+ stars:
[image]

80+ stars:
[image]

Stock AllSky startrails
[image]

I can post the cobbled modules (logger & sorter (sorter could have been just a little python script i guess..)) if anyone wants to test the method. 

Runs in module manager as the LAST module before save (to catch a little extra data). 
Grabs the metrics and writes them to a CSV file in ~/allsky/images/DATA/ (just a holding path, no specific reason -but the sorting script calls the file from the DATA folder). The output of the sorted images is generated into ~/allsky/tmp/ . 

For now though, I'm liking the output of the method. Starcount does seem to have some good promise in my testing. @EricClaeys helped by fixing up the logic in generateForDay.sh to enable the text-file list instead of a directory, as well as an updated startrails.cpp.  It requires the updated generateForDay to allow the txt file pass through.

EricClaeys, 2025-08-03:
@jcauthen78, nice improvement.

Just so I understand, you're using the "Star Count" module to get a star count and somehow weeding out the images with too few stars, then you pass that list of images to the `startrails --images file_name ...` command or to `generateForDay.sh --images file_name --startrails ...` ?

> The script still calls the threshold value, haven't figured out if there's a way to tell it to bypass the threshold

Which "script" ?  The `startrails` command?  If so, what I'm thinking is now that it accepts a file that contains a list of images, is to add the calculated mean brightness from the auto-exposure algorithm to each line in that file, so a every line would look like:
~~~
YYYYMMDD/image-20250710010203  0.234
~~~
When the `startrails` command see a number after the file name on a line, it would use THAT number when it compares the brightness to the Brightness Threshold, and not calculate its own value (which per prior discussions we know will be different due to the different ways the brightness is calculated).
Further, if the script that creates the list of files gets the "Brightness Threshold" from the settings file, it can add only images to the file that have a brightness less than the "Brightness Threshold".  That means every image in the file will be included in the startrails.

EricClaeys, 2025-08-03:
@jcauthen78,

> pass it into the endOfNight.sh

I'd vote to have the database query and creating the list of images that meet the criteria done in the "startrails" section of `generateForDay.sh`.  That way, the correct files are used even when someone manually calls `generateForDay.sh`.
If there are not database records for the specified day, then startrails creation would be done as it currently is, with no criteria other than Brightness Threshold.

EricClaeys, 2025-08-03:
@jcauthen78 

> Not sure how a % would work.

just thinking off the top of my head and ignoring the difficulty in implementing...

If we created an `allsky-config  get_average_starcounts` command, it could first check if there was any per-image data saved in the database that contained star counts.  If so, it would use those starcounts from one or more nights.  If not, it would look at some number of prior nighttime images and run starcount on them.  Looking at records in the database would be a gazillion times faster than looking at actual images to determine starcounts.  In theory, this should only be a factor right after someone upgrades to v2025.xx.xx, before it's had a chance to save image data to the database.
Or maybe, if there wasn't any database data yet, `allsky-config  get_average_starcounts` would simply tell the user to try again after a couple clear nights.  In the meantime, startrails creation would work like it does today, and use Cloud Cover module data if it existed. 

Either way, `allsky-config  get_average_starcounts` would then use those starcount(s) to determine an average number of stars.  Maybe it needs to also keep track of the average brightness of all the images to account for moon-less and moon-full nights.  Or at least track the moon visibility percent.
The average number(s) of stars would be saved to the database as well as the average brightness of the images used in calculating the average number of stars.  For example, it might come up with 120 average stars for a night when the average image brightness was 0.3 and 150 stars when the average brightness was 0.2.
As the user gets more nights of images, they could run `allsky-config  get_average_starcounts` again to get a better average number of stars under different conditions.

We'd add a new setting in the WebUI for the % of stars needed to include images in a startrails (in addition to the new setting for "Use nighttime only images" ??).
The % the user entered into the WebUI would be the percent of the average number of stars that need to be in an image.  The user wouldn't necessarily need to know what the average number is, but it probably wouldn't hurt to show it.
On a night when the average image brightness was 0.3 it would be a percent of 120 stars.
On a night when the average image brightness was 0.2 it would be a percent of 150 stars.

Entering a `0` for the percent would cause the starcount to be ignored.
We'd need to come up with a default percent - maybe 50% ?

EricClaeys, 2025-08-03:
@jcauthen78 
> Are you thinking the average data be generated with a Find my startrails settings button?

I'm finding the `allsky-config` command to be very useful and have been adding more commands to it every release, so at a minimum, I'd like to use it to do the guts of the work.
When a user enters `allsky-config` on the command line, they are shown a menu of possible actions, similar to `raspi-config`.  The action they choose may prompt for additional information if needed.  When the action completes, it writes its output to the screen, then shows the user the menu again.  Nice, but not useful to automate something.
However, if the user specifies an action to take on the command line, like `allsky-config  get_average_starcounts`, that action is executed with no prompting.  When the action completes it writes its output to the terminal and exits.  It's possible to add arguments to the action, like
~~~
allsky-config  get_average_starcounts --threshold 3.0 --images file_name
~~~
The beauty of this is that the WebUI could call the command, passing any arguments as needed, and the output can be displayed on the web page.
So we could also have a user-friendly web page where the user enters some options, then the WebUI passes that data to the command.  The real work is centralized in the command.

While developing this, we'd probably run the command on the command line.  Once we know what works and what data is needed, we could create the web front end (which we may delay until a point release).  I suppose it's possible that we don't need to prompt the user for much, and users only need to do this once, in which case we may not bother with a web front end.

Regarding WHEN this would be run, I think it would be really cool if the WebUI could figure out when it should be run, then prompt the user to run it.
If we want to only use data saved in the database, it should probably be run after a clear night or two.  The WebUI could also prompt the user to re-run it after a month (for example) of data.

EricClaeys, 2025-08-03:
@jcauthen78,
Your percents.txt file is impressive.
Eric

jcauthen78, 2025-08-03:
> [@jcauthen78](https://github.com/jcauthen78)
>
>
> I'm finding the `allsky-config` command to be very useful and have been adding more commands to it every release, so at a minimum, I'd like to use it [...]
I don't know enough about the allsky-config yet, so not sure what it can or can't do, I'll have to lean on your expertise on that end. 

> The beauty of this is that the WebUI could call the command, passing any arguments as needed, and the output can be displayed on the web page. So we could also have a user-friendly web page where [...]

That could work. It'll definitely need a fair average of nights to get initially tuned. If the user is lucky enough to have gorgeous skies 3 nights in a row, that'd probably be good enough. But a user like my situation, mostly cloudy for a few nights & smoky - mine is going to take a few more days to average out.  The idea of a monthly check could be helpful, maybe have the update once a month in the background, add the values to the DB, letting the user see what their skies did over the months - following the averages.  

> [@jcauthen78](https://github.com/jcauthen78), Your percents.txt file is impressive. Eric

Thanks :) The bortle is a massive W.A.G based on a lot of assumptions from googling random methods of determining bortle a few years back.  I wouldn't say its scientifically accurate - but was a fun test to throw in. The percentiles were cobbled from some of my other data analysis scripts from my weather station.  I'm a data hoarder, lots of random stuff all over the place.

craigim, 2025-08-04:
> If you were to prioritize all the improvements and pick just 1 or 2 for the next release and plan other improvements for follow-on releases, what would you pick? Keep in mind that we'll almost [...]

Having access to the metadata is pretty key right there, whether from the Exif or from a database because it opens up a lot more sophisticated processing and analysis. Before I would make big changes to the algorithm, I'd want to have an opportunity to play around with those variables in the processing to see what works the best.

> The RPi auto-exposure algorithm currently creates a mask internally that's (I think) a circle centered on the sensor and is 2/3 the height of the sensor.

That's basically what I was trying to recreate with my mask. I use 90% instead of 66%, but the idea is the same. My reading of the docs was that the whole image was used, but that could just be me skimming. In the python script I wrote, I made the size of the circle a command-line option with a default. Being able to point to a mask file would also be a good option.

> Are you willing to help with these improvements?

If the changes are in python, I can spend a little time on it, but that and MATLAB are my only two languages. I also exclusively use GitLab instead of GitHub at work, so there is going to be a bit of a learning curve there.

NiteRide, 2025-08-06:
Another thought for the backlog less technical than determining best images could be a feature/story to allow a user to apply an overlay mask at the end of creation to clean up the final image.  eg to remove all the blurry changing text (timestamp, sensor data, etc)? example:
[image]

_And maybe_  as a separate second sub-story to evaluate, possibly adding specific startrails image overlay text along with or separate from the mask.  Of course the variables I have included below add more complexity to the idea.  Maybe a user would want to add something like:
```
Startrails image from night of 2025-02-08. 
380 of 700 images used with an average of 85 stars detected.
```

[image]

EricClaeys, 2025-08-06:
@NiteRide,
Interesting idea.  Assume for discussion's sake that we added a new "startrails" flow in the Module Manager that had its own overlay template, and that flow was executed at the end of startrails execution.
Users could then add modules to the flow, like the Image Mask module followed by the Overlay module which would easily allow doing what you show.

v2025.xx.xx can now create multiple startrails or multiple stretched images using different settings (Brightness Threshold for startrails and Stretch Amount and Mid Point for stretches).  It then shows the images on one page so the user can decide which is best, and hence which settings to used.  Right now the file names have the setting values in their names and the file name is shown below each image.

I would like to add that info to the images, so I asked @Alex-developer about allowing a script to call a module or series of modules.  Exactly as described above, but for any new flow with any overlay template.  The script would create a test image, then create an overlay template with the setting values and appropriate text, then execute the flow with that template.  The flow would call the Overlay module which would add the text.  The script would then do the next test image.

Your suggestion and my request to Alex would both use the same mechanism, but with different parameters.

@Alex-developer,
Thoughts?