AllskyTeam/allsky #4916: v2025.xx.xx - several various bugs (as per 21-Sept version)

NiteRide, 2025-09-21:
### DO NOT USE!  For Allsky Team use only

### Problem:
a few bugs I picked up in v2025.xx.xx from 21-Sept test install

1. on **Allsky System > Change Password**: changing `Enable WebUI` from Yes to No or visa versa does not save.
No known workaround
2. in **Module Manager**: deleting (clicking trash can icon) to remove a module from a flow does not save.  returning back to the flow shows it back in selected modules.
Workaround: use the green << button to remove from flow and click the [Save] button at top of page
3. in **Module Manager > configuring a module pane**:   When including a mask,  after selecting a mask image, the page is no longer scroll-able while the module's config pane is shown.  So if there are more parameters they cannot be reviewed, and if the save button is off the bottom, the module config cannot be saved.  I discovered this while working on a new module.  I will provide the file separately if needed
Workaround: zoom browser page out until full module config pane is visible and save, then re-open module config if required.
4. Maybe not a "bug" but, if a module is not "enabled" in a flow, then the test button doesn't work.   Perhaps can a message be returned if a user clicks "Test" on a module that is not enabled so they know why the "test" didn't do anything?  Maybe something as simple as "This module must be enabled in the flow before a test can be executed"
5. "Ss" Typo in **Module Manager > Select Flow contextual help**
[image]
6. **Install > Upgrade** did not copy over previously created masks from allsky-OLD

NiteRide, 2025-09-22:
similar @Alex-developer, when i add charts, they don't save.  only in the browser session until i clear the cookies, etc.

Alex-developer, 2025-09-22:
> ### Problem:
>
> a few bugs I picked up in v2025.xx.xx from 21-Sept test install
>
> 1. on **Allsky System > Change Password**: changing `Enable WebUI` from Yes to No or visa versa does not save.
>    No [...]
  
Fixed. When a module is removed using the trash can then the save icon is enabled  
  
> 3. in **Module Manager > configuring a module pane**:   When including a mask,  after selecting a mask image, the page is no longer scroll-able while the module's config pane is shown.  So if there [...]

Do you have an example module that scrolls and also allows a mask to be added? I dont have one  
  
> 4. Maybe not a "bug" but, if a module is not "enabled" in a flow, then the test button doesn't work.   Perhaps can a message be returned if a user clicks "Test" on a module that is not enabled so [...]

Fixed. If the module is disabled then the test button is also disabled and a tooltip added explaining that you need to enable the module to test it  
  
> 5. "Ss" Typo in **Module Manager > Select Flow contextual help**
> 

Fixed